shaise/FreeCAD_SheetMetal #174: Fold Wall command fails if selections are made in the wrong order

jayennis22, 2021-01-31:
The Fold Wall command requires that you select a face and an edge.  Order of selection is important.  If you select the edge then the face, you will get an error as follows:

`19:37:13  Traceback (most recent call last):
  File "/Users/xxxx/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalFoldCmd.py", line 249, in execute
    s = smFold(bendR = fp.radius.Value, bendA = fp.angle.Value, flipped = fp.invert, unfold = fp.unfold, kfactor = fp.kfactor, bendlinesketch = fp.BendLine,
  File "/Users/xxxx/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalFoldCmd.py", line 98, in smFold
    normal = foldface.normalAt(0,0)
<class 'TypeError'>: function takes exactly 1 argument (2 given)`

You must select the face first, then the edge.

jaisekjames, 2021-02-04:
added here

https://github.com/shaise/FreeCAD_SheetMetal/pull/177

luzpaz, 2021-04-06:
@jayennis22 care to test and confirm?  
@shaise can we close this?

luzpaz, 2021-04-13:
> @jayennis22 care to test and confirm?
> @shaise can we close this?

ping

shaise, 2021-04-14:
waiting for @jayennis22

jayennis22, 2021-04-14:
The error condition as documented is fixed.  If you select the edge first and then the face, the Fold Wall icon is not enabled.  When you execute "Gui.runCommand('SMFoldWall',0)" in the Python console it fails silently.

I don't know what the design goals are for usability.  Perhaps this solution is considered sufficient.  However, I think it would be much more usable if SMFoldWall would instead accept an edge and a face in either order.  Failing that, SMFoldWall should at least provide a useful error message when the inputs are provided in the wrong order.

Leaving open for further review.  Thanks very much!

luzpaz, 2022-11-09:
Shall we close and open a new ticket or rename this one ?

shaise, 2022-11-10:
I think this can stay open, with added enhancement label

luzpaz, 2023-03-06:
@shaise perhaps renaming ticket for clarity ?