shaise/FreeCAD_SheetMetal #175: Fold Wall command allows you to select the body instead of a face (and fails)

jayennis22, 2021-01-31:
If you select the Body and an edge, the Fold Wall command  icon will become active and you can execute it.  The result will be an error as follows:

`19:56:37  Traceback (most recent call last):
  File "/Users/xxxx/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalFoldCmd.py", line 249, in execute
    s = smFold(bendR = fp.radius.Value, bendA = fp.angle.Value, flipped = fp.invert, unfold = fp.unfold, kfactor = fp.kfactor, bendlinesketch = fp.BendLine,
  File "/Users/xxxx/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalFoldCmd.py", line 96, in smFold
    foldface = FoldShape.getElement(selFaceNames[0])
<class 'IndexError'>: list index out of range

The Fold Wall command  icon should not become active unless a proper face object and an edge object are selected.
`

jaisekjames, 2021-02-04:
added here

https://github.com/shaise/FreeCAD_SheetMetal/pull/177

mads-leth-danielsen, 2021-02-26:
Fold Wall as in this video https://youtu.be/ZnKTVvaDflY dos not work in ver. 2,47 it workt in ver 2.43.... 
it would be nice if you didn't have to set the radius up your self... BG mads

closed 2021-02-28 by shaise