shaise/FreeCAD_SheetMetal #185: Uncorrect bend

sedatkrn, 2021-03-06:
freecad 019 and realthunder have got same problem about edge bend.
1- i make a rectangle 300x400 and then create pad from it. 
2- i made one edge to down  bend. edge  demations are 300mm. 
3-  all bend lenght is 20mm  
4- other 3 edge did bend upper.  but it is not coorect bend edge.

[image]
[image]

it is file :
https://drive.google.com/file/d/1S1tC_uXlMyu8vf47Rqs2GuqeWl5x61SG/view?usp=sharing

luzpaz, 2021-03-06:
cc @ceremcem

ceremcem, 2021-03-06:
@luzpaz Why me? :D 

@sedatkrn Somehow the `Invert` property of the feature `Bend` (the feature before `Bend0001`) haven't visually applied. I set the `Invert` property of `Bend` to `false` and then `true` again, it is fixed:

https://user-images.githubusercontent.com/6639874/110222416-0581ca00-7ee3-11eb-98e1-7ed689f86656.mp4

I guess it's related with LinkStage3 not with SheetMetal WB. I also deleted the bend features and created them from scratch, both in your version of LinkStage3 and the latest one as of today, both produced the correct bend.

ceremcem, 2021-03-06:
As for the intersection, same trick applies: 

https://user-images.githubusercontent.com/6639874/110222716-54c8fa00-7ee5-11eb-965d-f1ac8f46a618.mp4

```
OS: Debian GNU/Linux 10 (buster)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.27270 (Git)
Build type: Release
Branch: LinkStage3
Hash: 32f78691fae500e494879493abc2b74cff9585d3
Python version: 3.7.3
Qt version: 5.11.3
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

luzpaz, 2021-03-06:
> @luzpaz Why me? :D  

https://twitter.com/tc_ulgen/status/1368314503595757570  
:smirk:

sedatkrn, 2021-03-06:
Very interesting I try again. I made a strove whit freecad. The sheet metal wb extremely important for me. I must make many sheet panel, many bending. Some times the models could be very complicated. Thanks for interesting. Sagolasin bu konuyla ilgilenen bir Türk olması harika.

sedatkrn, 2021-03-07:
i tried whit freecad 018 last version but bend not work probably. i sening a video about issue
https://drive.google.com/file/d/13Cc-BC_9NRaEr-o44wyUqqT6ODI9SclC/view?usp=sharing

ceremcem, 2021-03-07:
You are selecting edges, not faces.

ceremcem, 2021-03-07:
https://user-images.githubusercontent.com/6639874/110237059-c71ff580-7f4a-11eb-8ad1-25966f4146b5.mp4

ceremcem, 2021-03-07:
> The sheet metal wb extremely important for me. I must make many sheet panel, many bending.

Btw, I strongly recommend enabling `Engineering UX Mode` for those who use this WB in production: 

[image]

sedatkrn, 2021-03-07:
Mu sheet metal version 2.47 yours 2.45

ceremcem, 2021-03-07:
Correct. @shaise I can verify that 2.45 has no problems where 2.47 causes above (original) issue.

luzpaz, 2021-03-07:
> > The sheet metal wb extremely important for me. I must make many sheet panel, many bending.
> 
> Btw, I strongly recommend enabling `Engineering UX Mode` for those who use this WB in production:

Maybe worth adding this screenshot to the README,md file?

ceremcem, 2021-03-07:
@shaise in v2.47, `Bend type: Material Inside` causes above issue. This bug is introduced by commit https://github.com/shaise/FreeCAD_SheetMetal/commit/f70b7d7.

ceremcem, 2021-03-07:
@sedatkrn As a workaround, you can use the commit before this bug is introduced: 

1. Go to `%APPDATA%\FreeCAD\Mod\`
2. There should be a `sheetmetal` folder. Rename it (like, `sheetmetal.backup`)
3. Download https://github.com/shaise/FreeCAD_SheetMetal/archive/9b5c3e10f6d4ee84754bfb2807c531eab70ac875.zip
4. Unzip it 
5. Now you should have `%APPDATA%\FreeCAD\Mod\sheetmetal` folder. 
6. Restart FreeCAD

(Sorry, I'm not a Windows user, so steps might not be extremely accurate)

shaise, 2021-03-07:
> 
> 
> @shaise in v2.47, `Bend type: Material Inside` causes above issue. This bug is introduced by commit [f70b7d7](https://github.com/shaise/FreeCAD_SheetMetal/commit/f70b7d7).

@jaisekjames , can you check this issue? It looks like it happened after your PRs 

shai

sedatkrn, 2021-03-07:
how i ccheck it? i add chnagged file from github?

sedatkrn, 2021-03-07:
working perfect  thanks :)
[image]

sedatkrn, 2021-03-07:
i tried linkstage and it is working correct :)

[image]

ceremcem, 2021-03-07:
Bug is not resolved, please reopen.

luzpaz, 2021-03-07:
@shaise :point_up_2:

sedatkrn, 2021-03-11:
is the bug still unrsolved?

ceremcem, 2021-03-11:
@sedatkrn Waiting for @jaisekjames to respond.

closed 2021-03-20 by shaise