shaise/FreeCAD_SheetMetal #186: Support for other materials (Cardboard, acrylic)

luzpaz, 2021-03-06:
Thread for reference: https://forum.freecadweb.org/viewtopic.php?f=24&t=49546

/cc @oficinerobotica

carribeiro, 2021-03-06:
The Sheet Metal Workbench is great, and can be extended to other materials. Cardboard is a great candidate but it behaves a bit differently than metal (or acrylic) and requires some adjustments. 

The main difference is that cardboard gets COMPRESSED inside the fold line, instead of STRETCHING on the outside. 

The way it works now, it's up to the user to make a bend with a larger radius. I believe the proper way is to have some option where the user can specify whether it's a "stretched" bend or a "compressed" bend, and the workbench calculates how big the radius should be.

Please note that IN MOST PRACTICAL CASES it does not make much sense to have a bend with a larger radius for cardboard, because (1) cardboard is usually used for boxes with sharp, 90º bends; and (2) even if you try cardboard is very hard to bend in an arbitrary radius unless you make the bend exactly parallel to the flutes of the cardboard's structure.

So - if the feature is deemed appropriate - the idea is:

1. have an option for stretched/compressed bends (stretched being the default for metal).
2. make the bend as usual, but considering that cardboard has a really short inner radius when properly folded. it may require some experimentation to set the proper default radius for compressed cardboard bends.
3. when unfolding, calculate how much material will compress inside the bend, and add this extra material to the unfolded part.

One related feature is https://github.com/shaise/FreeCAD_SheetMetal/issues/187: when unfolding, scoring lines have to be added. I REALLY DONT KNOW HOW TO MAKE IT :-) because the scoring line has to be added to the 2D design as a single line, not as a 3D object or part (more on the #187 issue).

ceremcem, 2021-03-06:
> The main difference is that cardboard gets COMPRESSED inside the fold line, instead of STRETCHING on the outside.

Which means `K-factor (ANSI)` of a cardboard is `1.0`, right?

carribeiro, 2021-03-07:
> > The main difference is that cardboard gets COMPRESSED inside the fold line, instead of STRETCHING on the outside.
 
> Which means `K-factor (ANSI)` of a cardboard is `1.0`, right?

I really had to read Wikipedia's entry to understand it better (I'm no mechanical engineer, just a maker with some technical background on IT & random stuff) :-) That said, https://en.wikipedia.org/wiki/Bending_(metalworking)#K-factor doesn't mention cardboard as a material - but I guess that you're right, as R is probably equal to T in the formula.

carribeiro, 2021-03-07:
Anyway, the question is that when unfolding you must add enough material to make up for the compression.

Also - I'd guess that there's a residual radius when folding and that the material on the outside will not necessarily follow an arc shape (it may be more similar to a straight line). Some testing may be required.

ceremcem, 2021-03-07:
> but I guess that you're right, as R is probably equal to T in the formula.

Nope, `t` is equal to `T` in the formula, thereby the K-factor (ANSI) is `t/T = 1`.  See https://github.com/shaise/FreeCAD_SheetMetal#terminology

> Also - I'd guess that there's a residual radius when folding and that the material on the outside will not necessarily follow an arc shape

I recommend the "measurement" approach: 

1. Get a rectangular shape of your material, say 50x30mm. 
```
               50mm 
----------------------------------
|                  :             |
|                  :             |  30mm 
----------------------------------
                   ^--- mold line (where the punch touches)
```
2. Bend it as you like. 
3. Scan it with a scanner, along with 2 rulers, one vertical and one horizontal (or place just another shape where you know exact dimensions). 
4. Calibrate the image: Open the scanned image, measure the rulers digitally and verify the rulers' lengths (and scale the image accordingly). 
5. Measure both of the `Flange`s (`Flange_1` and `Flange_2`). 
6. Measure the apparent radius. 
7. Calculate the apparent K-factor accordingly: https://ceremcem.github.io/sheetmetal_calculator/

> ```
> Flange_diff = (Flange_1 + Flange_2) - 50mm 
> ```

carribeiro, 2021-03-07:
I'll do it. It's now nearly 10pm :-) but I'm going to the shop tomorrow, I'll make it and put it here. I'll check if I have different grade of cardboard too.

luzpaz, 2021-04-27:
any progress?