shaise/FreeCAD_SheetMetal #187: Support for folding marks (for the laser cutter)

luzpaz, 2021-03-06:
reference: https://twitter.com/carribeiro/status/1368340185323352064

ceremcem, 2021-03-07:
I observed no benefit of marking the bend points with the laser cutter. Efficient approach (that I use) is 
1. Precisely calibrate the distance between punch and back gauge
2. Apply the correct force 

I obtain bends with 0.02mm of error with that approach. 

[image]

carribeiro, 2021-03-07:
This request is related to #186. Laser-cut cardboard boxes are easier to fold if the material is scored lightly with the laser cutter. It's usual to use a sparsely dotted line because a solid line weakens the cardboard. That's especially necessary when the fold is made against the direction of the flutes of the cardboard's structure.

As it is now, the workflow is:

1. design the box with the sheet metal workbench;
2. unfold the box;
3. generate a 2D drawing of the box, and export (SVG or DXF are the most common formats for lasers now);
4. open the drawing in a 2D tool of choice, and **ADD THE FOLDING LINES MANUALLY**.
5. cut the shape with the laser for folding.

NOTE: the 2D app of choice is usually CorelDraw, Autocad, or Inkscape. FreeCad itself or other open-source CAD could be used - but most laser cutting shops would probably end up using CorelDraw as far as I know for practical/historical reasons.

All this work is necessary because the scoring line is not a 3D part (although we could probably make a few thin holes with unfolding). In practice it's even a bit harder because some lasers (all that I know really - there may be exceptions) don't know how to deal with dotted lines and end up with a solid scoring line that weakens the cardboard.

I don't really have a plan on how to do it in FreeCAD but some possibilities are:

1. As said above, make thin holes on the fold line, and (somehow) convert the holes into a single stray line segment when generating the technical drawing or exporting the vector 
2. Investigate whether it's possible to generate the lines directly when generating the drawing.

carribeiro, 2021-03-07:
> I observed no benefit of marking the bend points with the laser cutter. Efficient approach (that I use) is
> 
> 1. Precisely calibrate the distance between punch and back gauge
> 2. Apply the correct force

I don't have the same equipment that you have to make folds. I do it by hand using aluminum L bars or MDF punchers cut to size. The folding lines make it easier to fold larger boxes. I know it may weaken the cardboard, but it's the most practical way to make it without requiring another equipment (specially for one-off custom boxes, or for prototypes).

HaasNL2, 2021-03-23:
OK I been making a box in sheetmetal today. And it goes decently, although it is hard to keep an overview. But once I have a box I can re-use it every-time.
I looked into professional box design software, and they are all just as expensive as normal CAD software. And for a small maker that makes a couple of boxes a year not viable. Like my boxes would cost 10 euro a box. I cannot charge 10 euro a for a box.

I currently design my boxes in Blender, but that has it cons. Like missing parametric design. Every-time I change my dimensions I have to scale stuff, which screw up my gaps and hooks that need to stay the same.
So doing this in Freecad would be way better. I also tried this in Fusion360 when I was still using that and it didn't work. At a moment Fusion just said screw this, this is to complicated I am bailing out. I noticed this also happens in Freecad to an extend. But I got further than Fusion so that is awesome! And probably with more work I can get a decent box design in it.
This is my design:
[image]
This is freecad when it had enough:
[image]

luzpaz, 2021-04-08:
anyone interested in spearheading this?

luzpaz, 2021-04-13:
@oficinerobotica can you weigh in since you made a cardboard box clip with LinkStage3

luzpaz, 2021-04-21:
@jaisekjames care to weigh in?

luzpaz, 2021-05-06:
@HaasNL2 are you planning to develop this feature?

HaasNL2, 2021-05-08:
@luzpaz 
To bad I am not a developer.
If you really want this it could be interesting to setup a gofund me and find a good developer that has the time to do this.
Or just donate to freecad and ask there.

carribeiro, 2021-05-08:
I'm still in the loop. At the moment have some real life things going on,
I`m moving my shop over this week, lots of things to do. But I think I'll
be back in a few days.

Carlos Ribeiro
Mentoria & Consultoria em Projetos de Inovação
about: https://about.me/carribeiro
mail: ***@***.***
twitter: http://twitter.com/carribeiro
medium: ***@***.***
linkedin: https://www.linkedin.com/in/carlos-ribeiro-6a7b8b1

hydroid7, 2026-08-15:
See https://github.com/shaise/FreeCAD_SheetMetal/pull/613

luzpaz, 2026-08-16:
@carribeiro heads-up

luzpaz, 2026-08-17:
@hydroid7 does 613 close this ticket completely or is there still needs left over ?

hydroid7, 2026-08-18:
So far I can judge, the complete feature is done. It is also tested in production by us.

shaise, 2026-08-19:
@hydroid7 , out of curiosity: What is this feature used for in your production?

hydroid7, 2026-08-19:
Needed bend thick material beyond the capabilities of break press. Making the lines manually was error prone.

ceremcem, 2026-08-20:
If you require a "bend line with back gauge distance mark on it", like " ---- 22.59 mm ---------- 22.59 mm ----", it becomes more useful for error detection.