shaise/FreeCAD_SheetMetal #192: Unfold error

mrclnz, 2021-04-08:
The unfold process fails no matter which face I select for the start. In fact from the error it seems it stumbles over 'something'. It says 

Engineering mode is not enabled.
14:25:50  Manual K-factor is being used: 0.15 (din)
14:25:50  Error Starting: invalid thickness at Face72
14:25:51  exception at line 2876("'NoneType' object has no attribute 'Edges'",)
14:25:51  ("'NoneType' object has no attribute 'Edges'",)
14:25:51  Traceback (most recent call last):
14:25:51    File "/home/mrclnz/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 2876, in accept
    foldLines = foldComp.Edges
14:25:51  AttributeError: 'NoneType' object has no attribute 'Edges'

which to me is the pythonish equivalent of a null pointer reference
Tried with the latest 24612 and 24267; can't try with the 0.18 series since the format changed

mrclnz, 2021-04-08:
[Coperchio NEBULA-COMBO.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/6278650/Coperchio.NEBULA-COMBO.zip)

Got issues uploading the file sorry

jaisekjames, 2021-04-11:
Pls change extent 1 & 2 from -1.0 to -1.1 for "Bend" in tree view. Currently inside small bend face is fused with grilled side face. We need a gap between faces. 

Due to this change face number changes, so other features depend on this may break. 

For bend feature, double click on tree view to change face selected.

mrclnz, 2021-04-13:
So it's like the issues with booleans and tangent faces…
Anyway, it worked, thanks

closed 2021-04-13 by mrclnz