shaise/FreeCAD_SheetMetal #193: Bend relief for welding

blessedbiped, 2021-04-09:
Feature request:
I'd like to add slots to the bends to ease folding and allow for edge welding (see picture). I currently do it by hand, but I think the unfolding macro should be able to add them directly.
[image]

luzpaz, 2021-05-06:
Can you also x-post this to the [FreeCAD forum sheetmetal thread](https://forum.freecadweb.org/viewtopic.php?f=10&t=11303)?  TIA

luzpaz, 2021-05-15:
Thread: https://forum.freecadweb.org/viewtopic.php?f=3&start=370&t=11303&sid=32206caac82970cbd2331e7d476490af#p502565

I encourage you to go in to more details to promote more discussion in the thread

alexneufeld, 2025-02-02:
this feature request is for perforated bends, which are now supported.
@luzpaz this can be closed

[image]

shaise, 2025-02-03:
Indeed this is supported now.

closed 2025-02-06 by luzpaz