shaise/FreeCAD_SheetMetal #198: `gap2` cuts wrong surface in simple "material inside bend"

mangelozzi, 2021-05-06:
For a simple bend with `Material inside bend` with a `gap1` and `gap2` set to 10mm (all is good):
[image]

Now change `gap2` to 10mm: (note the strange vertical cut that appears)
[image]

Simple test case file here: <https://forum.freecadweb.org/download/file.php?id=152437>

Using 
```
commit e64016fb930df5d57903b1c283ce3c473d3dfb27
Author: Shai Seger <shaise@gmail.com>
Date:   Thu May 6 19:29:42 2021 +0300
    add workbench terminology
```
With PR #197 applied, and FreeCAD version:
```
OS: Windows 10 Version 2004
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: releases/FreeCAD-0-19
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.6+
Qt version: 5.15.1
Coin version: 4.0.1
OCC version: 7.5.0
Locale: English/United Kingdom (en_GB)
```

shaise, 2021-05-07:
should be fixed with pr #199

mangelozzi, 2021-05-07:
> should be fixed with pr #199

Test file is working now, that was fast, thank you!!!

closed 2021-05-07 by mangelozzi