shaise/FreeCAD_SheetMetal #200: Using SheetMetal, PartDesign and Sketcher, after modifying sketch got error "cannot normalize null vector"

Dagobert2021, 2021-05-25:
I want to create a housing with holes. Therefore I used the the partdesign/sketcher.
Then, I used the metal sheet to add bends and extends.
After that, I modified the sketch holes and want to update it in partdesign.
However, some bends show error and the 3d-view does not get updated.

Software:
OS: macOS 10.16
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24276 (Git)
Build type: Release
Branch: (HEAD detached at 0.19.1)
Hash: a88db11e0a908f6e38f92bfc5187b13ebe470438
Python version: 3.8.8
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.4.0
Locale: C/Default (C)

It is all a fresh installation.

The shown error:
08:00:41  Traceback (most recent call last):
  File "/Users/harald/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalCmd.py", line 1096, in execute
    s, f = smBend(thk, bendR = fp.radius.Value, bendA = bendAList[i], miterA1 = fp.miterangle1.Value, miterA2 = fp.miterangle2.Value,  
  File "/Users/harald/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalCmd.py", line 684, in smBend
    mainlist, trimedgelist, nogaptrimedgelist = getBendetail(selFaceNames, MainObject, bendR, bendA, flipped, offset, gap1, gap2)
  File "/Users/harald/Library/Preferences/FreeCAD/Mod/sheetmetal/SheetMetalCmd.py", line 362, in getBendetail
    revAxisV.normalize()
<class 'Base.FreeCADError'>: Cannot normalize null vector

[image]

[2021-05-25_Gehaeuse.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/6536995/2021-05-25_Gehaeuse.FCStd.zip)

Dagobert2021, 2021-05-25:
On click of the red marked bend, this page opens. But after I press update, it does not solve the problem.

[image]

Dagobert2021, 2021-05-25:
I observed this problem twice. It seems to be always, since there are only two trials.
I have to remove all bends and extends and create them again to heal it.
After modifying the sketch and update it, the problem occurs again.

Dagobert2021, 2021-05-25:
After update of bends, it sometimes shows all bends and extends, but with the old version of the sketch (missing holes).
[image]

Dagobert2021, 2021-05-25:
After load the project again in FreeCAD, all holes of sketch visible, but I miss some bends/extends.

What could be the root cause?

Is it my MacOS?
Actually, it is a simple drawing.
How to avoid this?

shaise, 2021-05-27:
The root cause is how freecad handles naming of features of solids. When you set a bend, freecad assign this bend to a feature, say "Edge 56". The problem is if you go back and make changes to earlier steps in the solid, freecad might rename all features to other names. this cause "Edge 56" to represent a new feature, and the bend will not work. The only way to overcome it as of now, is, after making the alterations that messed the object, go step by step to all steps in the body marked "!", double click on them, and reselect the correct edge.

shai

Dagobert2021, 2021-07-13:
If I walk through the list of parts, there is always only one edge to select and to reassign.

shaise, 2021-07-14:
you need to select the new edge(s) on the model, then click the update button on the left to assign the new edges

luzpaz, 2023-03-19:
@shaise can we close this ticket ?

shaise, 2023-03-20:
Yes, I guess so.