shaise/FreeCAD_SheetMetal #202: Parametric unfold sketch

federicofraysse, 2021-06-17:
Good Morning,
First of all, thank you for the work you have done.
I would like to know if it is possible for the unfolded sketch to be updated with the changes in the model. This is very important, since by taking advantage of the spreadsheets, many stamped profiles of different sizes can be designed, and thus be able to reuse the cutting and stamping drawings. The drawings should automatically update when you change the part configuration. I hope I explained well, Federico from Argentina.

alexneufeld, 2025-01-31:
This issue describes the behavior of the current unfolder exactly (within the limits of toponame tracking of course). This can be closed @shaise

closed 2025-01-31 by luzpaz

shaise, 2025-02-02:
@luzpaz , thanks, and sorry for the late reply.