shaise/FreeCAD_SheetMetal #206: Color of object changes on every step

berberic2, 2021-07-04:
If in [PartDesign preferences](https://wiki.freecadweb.org/PartDesign_Preferences) the option _Shape color Random_ is set, every sheetmetal operation changes the color of the object.

Syres916, 2022-08-04:
See https://github.com/shaise/FreeCAD_SheetMetal/pull/246 At present it won't handle individual face colours but it does handle shape colour and transparency

luzpaz, 2022-08-05:
Does anything on the PartDesign side need to happen?  Or is this just solely SheetMetal related ?

shaise, 2022-08-05:
It looks like that in part design adding features does not alter the color (the random color is selected only when creating the first feature). After the fix, sheetmetal behaves the same.

Syres916, 2022-08-05:
Yep, if the OP wants each Bend/Extend etc operation to look up the Random Color checkbox in Preferences of Part Design and apply something like `obj.ViewObject.ShapeColor=(random.random(),random.random(),random.random())` then I believe it can be done but that's not how I read the report.

luzpaz, 2024-09-09:
ping @berberic2

berberic2, 2024-09-09:
> ping @berberic2

What?
Problem still exists partially. Deleting a SheetMetal operations changes color of object.

shaise, 2024-09-09:
Yes, it was fixed and broken again due to the new material system. Need to see how to adapt to it.

berberic2, 2024-09-10:
Oh. I now have the problem not only in SheetMetall but in PartDesign too… 
The new material-system still seems to need a lot of work…

luzpaz, 2024-09-10:
Can someone link the upstream freecad ticket related to this ?

berberic2, 2024-09-10:
Showing:
[colorfull.webm](https://github.com/user-attachments/assets/aedd1391-104b-4992-a7f9-caa75b8ee795)

```
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38691 +4 (Git)
Build type: Release
Branch: Meins
Hash: 597ab8282e9c772ba7e7c32d5762087b7635fbc9
Python 3.11.2, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: /FreeCAD Classic/
```