shaise/FreeCAD_SheetMetal #214: No relief on added wall

protyposis, 2021-11-09:
Please consider the following basic example:

 1. Create base plate / additive box (Part Design) 10x10x1mm
 2. Select any box edge
 3. Use [`AddWall`](https://wiki.freecadweb.org/SheetMetal_AddWall) command
 4. Set `gap1` and `extend1` to 2mm in the properties of the added `Bend`
 5. See result from screenshot below

[image]

It seems to me that the workbench adds reliefs only to the side where a bend originates. Is it expected that there is no relief on the added wall (marked in green)? What is the recommended approach to get reliefs on both walls? (I do not mean a second relief on the base plate, as created when setting `gap2`).

kotarou3, 2025-09-27:
+1 to this feature. It still doesn't exist in the latest version.
Having reliefs only on one side is rather unusual, especially since a press brake bends both sides symmetrically

Ideally, we would be able to choose different relief parameters for the two sides of the bend as well

[image]

[example.zip](https://github.com/user-attachments/files/22577712/example.zip)

shaise, 2025-09-28:
Sounds reasonable.