shaise/FreeCAD_SheetMetal #216: automatic rebuild unfold update Ttechdraw

mads-leth-danielsen, 2021-12-07:
it wood bee nice if you could set it to automatic update unfold plate...   automatic update unfold plate  Techdraw

Best regard mads leth danielsen DK

https://www.youtube.com/channel/UCpdgfEnmKNi8iv_A66efg9Q

luzpaz, 2022-11-09:
@mads-leth-danielsen please be more specific

rola22, 2023-01-29:
If i understood correctly @mads-leth-danielsen want to somehow link unfolded solid to base model, so if you for example make a hole in base model it will automatically appear in unfolded solid and corresponding sketches with bend lines.

Sorry for bad english but i believe you know what i mean :)

EstebanFPV, 2024-11-27:
I think the point of this is so if you make a change to the object, it automatically re-unfolds it or otherwise updates the unfolded part somehow, so you don't have to reunfold it to export dxf's and add all your views and dimentioned in tech draw bend detail manually again.

alexneufeld, 2025-01-31:
I think this can be closed. Unfolding is now fully parametric and automatic to the same extent as any other solid modelling feature inf FreeCAD

luzpaz, 2025-02-01:
heads up @mads-leth-danielsen  
Closing.

closed 2025-02-01 by luzpaz