shaise/FreeCAD_SheetMetal #222: Unfolding Problem round parts

wwied, 2022-01-20:
first of all i apologize for my bad english - it's not my native language.
i've tried to unfold a circular bended part. works without any problems, part can be generated by different methods and workbenchens.
if i add cirular holes, unfold only works if the hole is not in the seperation line of the bending.

[image]

you can see here geometry that works for unfolding. if the cutout is extended to the opposite side, unfold is no longer possible.

and help would be appreciated!

shaise, 2022-01-21:
Unfortunately making holes on a round part is not supported well with the unfolder. only holes on flat parts.

luzpaz, 2022-11-09:
Should we turn this in to a feature request ? Or just add it as a wont fix limitation ? Is it technically feasible to do with Sheetmetal wb ?

luzpaz, 2022-11-09:
Dupe of #225

closed 2022-11-09 by luzpaz