shaise/FreeCAD_SheetMetal #223: SheetMetal disappears from the AddonManager after workbench update

ifohancroft, 2022-01-27:
I went to the Addon Manager and saw there are updates for KiCADStepUP and Sheet Metal. I've updated them. Then, after updating the Sheet Metal workbench, I clicked check for updates and it found one more update. I installed it as well and restarted FreeCAD. Now the Sheet Metal workbench is missing from the Addon Manager.

Here's FreeCAD's details:

```
OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.26683 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0388fbc)
Hash: 0388fbc98d49d874fb341b9037a743bc691d501f
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)
```

shaise, 2022-01-27:
I can't reproduce this issue. Can you try the latest freecad build?

ifohancroft, 2022-01-27:
Unfortunately, I cannot currently try with the latest version. I am having trouble compiling FreeCAD without it being all mangled so I am using the weekly builds' AppImage, however with the latest AppImage the AddonManager can't currently reach GitHub.

I have decided to try uninstalling the SheetMetal workbench on the older AppImage to see if it would show up in the AddonManager (I also hit Referesh local cache) but it didn't show up.

Since you can't reproduce it, I am guessing it's some bug in that particular FreeCAD version. I am closing the bug for now and when I can access the AddonManager on the latest version, if the bug shows up again, I'll report back.

ifohancroft, 2022-01-27:
Ok, so I managed to get the AddonManager working by changing its settings from 'User-defined proxy and empty value for the proxy' to 'No proxy' and the problem persists. Sheet Metal is not there. Even if I go in Preferences and uncheck 'Hide addons marked as obsolete' and 'Hide addons marked as python2'

ifohancroft, 2022-01-27:
Ugh.. sorry for the multiple contradicting comments. Now hitting Refresh local cache seems to work and it shows up again, so once again I am closing the issue.

Here's the FreeCAD info:

OS: Manjaro Linux (MATE/mate)
Word size of FreeCAD: 64-bit
Version: 0.20.27078 (Git) AppImage
Build type: Release
Branch: (HEAD detached at bcdd926)
Hash: bcdd926686f2245fe60baface1cce1ab3dff7a5b
Python version: 3.9.9
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

closed 2022-01-27 by ifohancroft

shaise, 2022-01-28:
The last change to the workbench was adding a package.xml meta-data file to be used with the new Addon Manager system. (As I was asked by the freecad team) So I guess it had something to do with it. Perhaps the new system still needs some ironing.