shaise/FreeCAD_SheetMetal #225: Part with chamfered hole wont unfold

kemes, 2022-03-05:
A simple part with a chamfered hole can't be unfolded. As the top face selected when you try to unfold an error occurs saying: _Analysis: Sheet thickness invalid for this face!_ A failed face is created to the tree.

[image]

shaise, 2022-03-05:
Unfolder support only "Legal" sheets -  those that after unfolding have only 90 degree edges (and hence can be laser cut)

kemes, 2022-03-05:
Any plans to support chamfered holes?

shaise, 2022-03-06:
The unfolder tool is actually en external macro that was embedded into the WB. It is not regularly maintained. This code is quite complicated as it is, and adding non 90 degrees edges makes it far more complicated. So unless someone new decides to dive into this code, I don't see this feature being implemented any time soon.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise