shaise/FreeCAD_SheetMetal #227: Using SheetMetal with FreeCAD without GUI (only CLI)

gabrielgriesser, 2022-04-11:
Hello,

Is it possible to use the SheetMetal module in a python3 script which uses only FreeCAD cli commands ?

In fact, I would like to use the **SheetMetalUnfolder** function, but it seems FreeCADGUI has to be imported.

Can I unfold some shape without using GUI ?

Thank you in advance :)

shaise, 2022-04-11:
the sheet metal unfolder is an external macro that was integrated into sheetmetal workbench. It uses freecadgui. So, without code changes I don't think it is possible.

gabrielgriesser, 2022-04-11:
> 

Okay thank you for the quick answer ! :) 
So I should be able to use FreeCAD SheetMetal in a **Dockerized FreeCAD with GUI container** ? (for ex. https://github.com/ToddG/freecad-builder or https://gitlab.com/daviddaish/freecad_docker_env).

shaise, 2022-04-11:
Not familiar with that, so I really cant say.

luzpaz, 2022-04-11:
@gabrielgriesser feel free to post on the freecad forum. There is an interest in running FreeCAD in 'headless' mode and this could make a nice usecase.

luzpaz, 2022-04-26:
?

gabrielgriesser, 2022-04-27:
Sorry, here is the thread on the freecad forum !
https://forum.freecadweb.org/viewtopic.php?f=9&t=68323

zuchtzucht, 2023-03-17:
Any update on this ?