shaise/FreeCAD_SheetMetal #235: Sheetmetal V50 not giving sketch on unfold

Reidwkw, 2022-07-17:
I noticed that V50 had an update that its using techdraw for the sketches on the unfold feature. I went to another laptop with the v49 and it is properly giving the unfold sketch. so i know its a v50 issue especially because the error talks about "techdraw has no attribute". here is the error code i am getting in the report view after using unfold with v50
16:17:13  Manual K-factor is being used: 0.40 (ansi)
16:17:13  Traceback (most recent call last):
  File "C:\.....\FreeCAD\Mod\sheetmetal\SheetMetalUnfolder.py", line 2899, in accept
    grp1 = TechDraw.projectEx(s,norm)
<class 'AttributeError'>: module 'TechDraw' has no attribute 'projectEx'

until this gets fixed...is there a way to revert back to v49? id rather not use my work laptop which is what has the v49 that is working

luzpaz, 2022-07-17:
Can you provide your full [About](https://wiki.freecad.org/About) info, please?

Reidwkw, 2022-07-17:
Here is the copied About
OS: Windows 10 Version 2009
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24267 +148 (Git)
Build type: Release
Branch: Branch_0.19.4
Hash: 476ecf091941bead59b14e44afa6064d5a66afa3
Python version: 3.8.6+
Qt version: 5.15.2
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)

shaise, 2022-07-18:
it probably has something to to with pr #234 by @easyw .

Reidwkw, 2022-07-18:
> @Reidwkw <https://github.com/Reidwkw> are you planning to move to v0.20
> soon?

Would updating to v0.20 fix the issue or are we just trying to see if the
issue persists in the updated version?

akshimassar, 2022-07-27:
```
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 2022.709.26244 +5001 (Git) AppImage
Build type: Release
Branch: LinkDaily
Hash: 096210d21183e9dfdc3b25777760bfb6c00a210b
Python version: 3.9.13
Qt version: 5.12.9
Coin version: 4.0.1
OCC version: 7.5.3
Locale: English/United States (en_US)
```
Same problem for linkstage branch

Syres916, 2022-10-23:
See https://github.com/shaise/FreeCAD_SheetMetal/pull/251

luzpaz, 2022-11-09:
What's the status on this ticket?

luzpaz, 2023-03-06:
@Reidwkw ping

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise