shaise/FreeCAD_SheetMetal #243: "Make Wall" and "Extend Face" tools don't work

GS90, 2022-07-29:
Hello everyone, thanks for your work.

After the latest changes (pull request #242), some tools have errors.
For example, when using "Make Wall" I get the following message:

```
19:58:24  Traceback (most recent call last):
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 1116, in execute
    s, f = smBend(thk, bendR = fp.radius.Value, bendA = bendAList[i], miterA1 = fp.miterangle1.Value, miterA2 = fp.miterangle2.Value,
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 698, in smBend
    mainlist, trimedgelist, nogaptrimedgelist = getBendetail(selFaceNames, MainObject, bendR, bendA, flipped, offset, gap1, gap2)
  File "/home/user/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalCmd.py", line 364, in getBendetail
    pThkDir2 = lenEdge.Curve.projectPoint(pThkDir1, "NearestPoint")
<class 'AttributeError'>: 'Part.Line' object has no attribute 'projectPoint'
19:58:24  Recompute failed! Please check report view.
```
My steps are simple:
new body -> new sketch -> rectangle -> pad (thickness 2mm) -> select edge or face, "make wall" -> error.

FreeCAD info:
```
OS: Ubuntu 22.04 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.29177 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20)
Hash: 68e337670e227889217652ddac593c93b5e8dc94
Python 3.9.13, Qt 5.12.9, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.53
```

GS90, 2022-07-29:
In version 0.21.29485, the latest available on "weekly-builds", this error does not occur.

shaise, 2022-07-30:
Hi all.
Confirmed. It seems that @Roy-043 fix has broken basic workflow on stable version 20.
I have added a temporary fix. Please update and let me know if it fixes your issues

shai

GS90, 2022-07-30:
Yes, after the update the tools work in 0.20 and 0.21.
Thanks.

closed 2022-07-30 by GS90

Roy-043, 2022-07-30:
Sorry to all for introducing this problem.
https://github.com/shaise/FreeCAD_SheetMetal/pull/244 should fix things.