shaise/FreeCAD_SheetMetal #247: Translation of GUi

kaktusus, 2022-08-08:
How likely is this workbench to join the FreeCAD-addons on the Crowdin platform?

I am interested in the author's opinion in this regard.

shaise, 2022-08-08:
Sorry, not familiar with Crowdin

kaktusus, 2022-08-08:
[Kuzma30](https://github.com/Kuzma30) Are you able to help a colleague and explain the most important points to him?

kaktusus, 2023-05-24:
A developer's manual was created:

https://github.com/FreeCAD/DevelopersHandbook/blob/master/technical/translation.md

hasecilu, 2024-03-09:
From what I have seen on the main FreeCAD Crowdin project almost every day FreeCAD is being translated, on the addons project only 7 external WB are there from more than 100!.
The point is that having the same people that translates main FreeCAD translating the addons will help end users by having a level of consistency on the whole FreeCAD environment.

I'm sure doing it will get more attention from translators.

luzpaz, 2024-09-10:
I think this can be resolved due to https://github.com/shaise/FreeCAD_SheetMetal/blob/master/TRANSLATIONS.md

closed 2024-09-10 by luzpaz