shaise/FreeCAD_SheetMetal #249: Accessing SheetMetal Workbench through API

stgeorges, 2022-09-18:
Hello,

First of all, thank you for developing and releasing for free SheetMetal Workbench. You basically allowed us professional sheet metal software users to save thousands per year. Thank you on that.

I would like to know: is it possible to access SheetMetal Workbench through FreeCAD API? Or does FreeCAD API not cover any Workbench?

shaise, 2022-09-19:
Hi @stgeorges ,
As far as I know it is possible in general to access FC workbenches through the API but it gets complicated when the WB is too tied to the GUI. I am not all that familiar with the FreeCAD API and never used it, so I can not say if SheetMetal meets all the conditions. It is best to ask about in on the FreeCad forum.

shai

stgeorges, 2022-09-19:
Hi @shaise ,
Thank you for the reply.
How was SheetMetal Workbench created then? Not through FreeCAD API, but instead through FreeCAD Macros?

shaise, 2022-09-19:
It depends what you consider as freedad api. everything uses freecad api, both macros and workbenches. But usually when people talk about freecad api they mean accessing the api offline without loading the freecad GUI. In this case it works with workbenches/macros that can work without GUI.
What do you need from the API?

stgeorges, 2022-09-19:
Hi @shaise,
Thank you once again for the reply.
What I am trying to achieve is call SheetMetal Workbench functionalities from outside of FreeCAD, regardless if FreeCAD is running or not. It would be better if FreeCAD is not running, but if it is, that would also not be an issue. I would create an Excel sheet inputs, or WindowsForms window, where certain inputs will be provided (Width, Height, sheet thickness (1,2,3mm...) etc) and based on those inputs, a 3D and .dxf files of a box-like bent sheets will be created from FreeCAD.

shaise, 2022-09-19:
Ok, Then indeed you want to call FreeCad externally. As said I have no experience with that so I cant say if this will work with sheetmetal. I do know that if a certain workbench requires GUI to perform tasks then accessing these functions outside freecad will probably not work.  
I do recommend trying your luck on the freecad forums, as I know some users did try to work with sheetmetal outside freecad.

stgeorges, 2022-09-19:
Thank you once again for the reply @shaise . I will do as you said.

luzpaz, 2022-09-19:
Closing ticket for now. 
@stgeorges for posterity would you mindnx-posting your forum thread link here so others can follow your inquiry?

closed 2022-09-19 by luzpaz