shaise/FreeCAD_SheetMetal #257: [enhancement] - fold radius

rola22, 2022-11-29:
Hello.

Right now if we fold a wall, we have to change fold radius for each fold. It's tedious if we make a lot of separate folds.
What do you think about making radius value of first fold the main value, which would be inherited by every next fold + add another property which will allow to override this value?

Sorry for poor english :)

jurpoc, 2023-01-23:
Hello,
I would have the same request, if possible. This would make our life much easier ;-)
Thanks a lot.

luzpaz, 2023-01-23:
@alexneufeld any interest in tackling this ?

alexneufeld, 2023-01-24:
Sure.

jurpoc, 2023-01-24:
Sorry, if I am asking too much, but would it be possible to set the default radius dimension from the first feature used wether it is a _Fold_ or _Make wall_ or _Make base wall_.
Then, whenever a new feature is added in the model, it will have the default radius from the first feature, but this could still be changed in the properties of single feature independently afterwards, if needed.

Again, sorry for asking for more work, but this would be very appreciated.

shaise, 2023-01-24:
I think the simplest solution is to always remember the last set radius and use it as the default for the next folds

jurpoc, 2023-01-24:
Yes, this would be perfect. It would be wonderful, if this would work on all features that "bend material" and have radius parameter.
Thanks a lot.

shaise, 2023-01-24:
Please update and see if this fix helps

jurpoc, 2023-01-25:
I tested the code.
1. When "Auto Link Bend Radius" is disabled, it works as it did before, as expected.
2. When enabled:
- It works in a way that every new Bend (Make Wall feature) gets the default radius parameter from the first Bend or BaseBend. If you set a different radius for the first Bend or BaseBend, all features are updated. Which is in my opinion exactly as it should work. 
- Unfortunately it does not work with Fold (Fold a Wall) feature. If this feature is the first in the design tree, the next ones are not linked. If this feature is not first in the design tree, this feature radius is not linked to the first.

Would it be possible to add the same functionality from the BaseBend and Bend also to the Fold?

Thanks a lot.

shaise, 2023-01-25:
hi @jurpoc 
It seems that you are working on @alexneufeld solution.
My solution is different, without the base bend link option.
please update from the main version here.

jurpoc, 2023-01-25:
Hi @shaise 
yes, the above remarks are for @alexneufeld solution.

Now, I tested also your proposal with this results:
- It works in a way that every new _Bend_ (_Make Wall_ feature) gets the radius parameter from the previous _Bend_ or _BaseBend_. 
- Unfortunately it does not work with _Fold_ (_Fold a Wall_) feature, as it does not the @alexneufeld solution. It would be very welcome if it did.
- Another unexpected behavior is that if there is an open document with a _Bend_ and a new document is created the first _Bend_ of the new document gets the radius from the previous document, while if you create a _BaseBend_ in the new document the radius value is always 1mm.

Maybe my reasoning is limited by how SolidWoks implemented this feature, but I think the @alexneufeld solution is more appropriate from the user point of view. At least in my opinion this solutions gives a better work flow.

Anyhow, any of the proposed solutions would be a huge improvement in Sheet Metal work flow. Even more, if we can extend this functionality to _Fold a Wall_ feature.

shaise, 2023-01-25:
OK try now,
It should work across all bending commands

jurpoc, 2023-01-25:
Yes, it works across all bending commands, but the behavior is not very intuitive for the user.

I made a document with a Body -> BaseBend [r=1] -> Bend [manually set r to 0.6].
Then I made a second document with a Body -> BaseBend [r=1 (default)] -> Bend [r=0.6 (default retrieved from the first document)] -> Bend001 [manually set r to 0.2]

I closed all the documents. Made a new document with a Body -> BaseBend [r=1 default] -> Bend [r=0.2 (default retrieved from the last changed radius in an already closed document].

OK, the above work flow is very unlikely to happen, but from an user perspective it is better not to have such a solution, because the probability to make a mistake during the design of the parts is higher in respect to the current solution of the radius always set to 1mm.

On the other hand, the solution propose by @alexneufeld has at least three advantages:
- is contained within a single document, 
- user can always check the radius setting in a single point (the first bending feature)
- user can always change the radius setting in a single point (the first bending feature)

All above points minimize the probability of making a design mistake and setting the radius of a single bending feature wrong, which leads to producing a part that is not consistent with the model and cannot be used. So the model needs to be rechecked and another part needs to be produced (waist of time and money).

I apologize for being so meticulous, but it is better not to make a change that would lead to a worse result than the current situation.

shaise, 2023-01-25:
So, If I understand correctly, You prefer alex's solution, only thing is to implement it as well across all bending commands.

jurpoc, 2023-01-26:
Yes, correct.

closed 2023-01-26 by shaise

shaise, 2023-01-26:
OK, Please try now. Alex's solution now works across all bend commands.

jurpoc, 2023-01-26:
Yes, it works perfectly. Great job.
Thank you very much. 
Let me know, if I can be of some help (apart from a modest contribution to FC account I did recently).

luzpaz, 2023-01-27:
Perhaps update the documentation?

jurpoc, 2023-01-31:
Sure, just let me know where and how to access the edit mode. Part of the documentation I was reading is [here](https://wiki.freecad.org/SheetMetal_Workbench) and some other [here](https://github.com/shaise/FreeCAD_SheetMetal).

luzpaz, 2023-01-31:
@shaise where do you recommend the documentation be updated?

shaise, 2023-02-01:
I think all the how-to-use docs should be in the freecad wiki. I will add a lint to it in the README.

jurpoc, 2023-02-02:
OK. I requested the read access to wiki. Then I need some time to study how to structure the documentation according to wiki requirements and I will start.
I can edit only the English version of the docs and read the Italian part, but I am not fluent enough in Italian to write in it.

PR #263 (2023-01-26): Add the option to auto-link wall bend radius
Adds a new preferences entry to automatically link the bend radius of newly created sheet metal walls to their earliest dependent Bend or BaseBend object.

Closes #257

[image]

[image]

This is a first attempt, I would appreciate comments on the implementation and design of this feature, as well as testing to see if it breaks in any edge cases.