shaise/FreeCAD_SheetMetal #259: Adding sketch for duplication breaks WallForming

FlorianUekermann, 2022-12-28:
I designed a forming tool, which I use on a plane with the WallForming feature. This works great. However, as soon as I add a sketch with two circles (to position and duplicate) the feature will be present in the expected places, but there is no hole in the sheet.
On a recalculate, I see the following two warnings: 
```
17:46:17  Sketcher::SketchObject: Link(s) to object(s) 'XY_Plane099' go out of the allowed scope 'Sketch008'. Instead, the linked object(s) reside within 'Body005'.
17:46:17  PartDesign::FeaturePython: Link(s) to object(s) 'Fillet003 Sketch008' go out of the allowed scope 'WallForming'. Instead, the linked object(s) reside within 'Body005 Body005'.
```
An example project demonstrating the problem is [attached](https://github.com/shaise/FreeCAD_SheetMetal/files/10315408/example.zip). 

To "fix" the example remove `Sketch008` from the WallForming element. (The result is the expected feature with matching hole in the middle of the sheet.)

shaise, 2023-01-05:
It looks like Sketch008 is not attached to the correct face.
I created a new sketch, attached to Pad001:Face21 (the face where the punch should go through) and changed WallForming to use this corrected sketch.
See attached.
[example_fixed.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/10352640/example_fixed.zip)

luzpaz, 2023-01-05:
Is this user error then? Can we close ticket?

FlorianUekermann, 2023-01-05:
Yes, this is user error. Closing.

However, it would be great to document this a little more clearly. On https://wiki.freecadweb.org/SheetMetal_Forming it only says must be coplanar to embossed surface. Only after seeing the fixed version I understood what to set as support (I didn't even know it is possible to set faces as support, because I'm a noob :-), so it isn't something I tried).

luzpaz, 2023-01-05:
@FlorianUekermann can you [DM](https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=12229) me your email and preferred username and you can add that to the wiki yourself?

FlorianUekermann, 2023-01-05:
@luzpaz 
Just registered: https://forum.freecadweb.org/memberlist.php?mode=viewprofile&u=56907
I can't send DMs yet due to the account being new.

luzpaz, 2023-01-05:
Create a post so I can give you posting perms

FlorianUekermann, 2023-01-05:
> Create a post so I can give you posting perms

Done: https://forum.freecadweb.org/viewtopic.php?f=3&t=75043