shaise/FreeCAD_SheetMetal #260: "Error Starting: invalid thickness" when unfolding

nicocesar, 2023-01-04:
I'm using FreeCAD 0.20 Revision: 14555 (Git shallow) released 23.12.2022  under Linux with  sheet metal add-on V0.2.60

When I did unfold I get this output in the console: 

```
Engineering mode is not enabled.
Manual K-factor is being used: 0.40 (ansi)
Error Starting: invalid thickness at Face4
exception at line 3298("'NoneType' object has no attribute 'Edges'",)
("'NoneType' object has no attribute 'Edges'",)
Traceback (most recent call last):
  File "/home/nico/snap/freecad-realthunder/common/.FreeCAD/Mod/sheetmetal/SheetMetalUnfolder.py", line 3298, in accept
    foldLines = foldComp.Edges
AttributeError: 'NoneType' object has no attribute 'Edges'

```

with a generic "Error Starting: invalid thickness at Face4"

The file is attached. 

Could this be a consequence of adding a formula as the thickness(Lenght) of the pad?  

[unfold_error.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/10347848/unfold_error.zip)

nicocesar, 2023-01-05:
One extra datapoint: It seems formulas expressions like "a * b" make this failure.

shaise, 2023-01-05:
The reason the unfolding failed is because the lips of the side folds touches the side wall with no spacing. This overlap of faces sometimes causes freecad to fail some operations (such as boolean operations that are used with sheetmetal)
If you change the thickness to 1.99mm or change other dimensions to prevent this overlap, the folding will succeed.

luzpaz, 2023-01-05:
Closing ticket

closed 2023-01-05 by luzpaz