shaise/FreeCAD_SheetMetal #262: Fold a Wall operator fails with "list index out of range"

johschmitz, 2023-01-23:
I am running FreeCAD v0.20.2 with SheetMetal Workbench v0.2.60.
When I try to use the Fold a Wall operator it fails with the following error message:

```
21:32:41  1.0
21:32:41  Traceback (most recent call last):
  File "/home/schmitz/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalFoldCmd.py", line 260, in execute
    s = smFold(bendR = fp.radius.Value, bendA = fp.angle.Value, flipped = fp.invert, unfold = fp.unfold, kfactor = fp.kfactor, bendlinesketch = fp.BendLine,
  File "/home/schmitz/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalFoldCmd.py", line 155, in smFold
    solid0 = solid1list[0]
<class 'IndexError'>: list index out of range
21:32:41  Recompute failed! Please check report view.
```

Here a screenshots to show what I did just before clicking the button:
[image]

jaisekjames, 2023-01-24:
folding line should on selected face. It look like folding line on opposite face.

johschmitz, 2023-01-24:
Can it not fold in both directions depending on which side the line is located?