shaise/FreeCAD_SheetMetal #265: CornerRelief bug : local variable 'BendEdge' referenced before assignment

nm2107, 2023-03-10:
Hello :)

Thank you for the workbench !

I'm running into an issue when trying to create a bent box. I can make the 4 bends at once by selecting the 4 sides of the base plate, but when I try to make a CornerRelief on one corner, I have the following error : 

```
19:01:24  Traceback (most recent call last):
  File "/home/<username>/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/sheetmetal/./SheetMetalCornerReliefCmd.py", line 448, in execute
    s = smCornerR(reliefsketch = fp.ReliefSketch, ratio = fp.SizeRatio, size = fp.Size.Value, kfactor = fp.kfactor,
  File "/home/<username>/.var/app/org.freecadweb.FreeCAD/data/FreeCAD/Mod/sheetmetal/./SheetMetalCornerReliefCmd.py", line 321, in smCornerR
    facelist = resultSolid.ancestorsOfType(BendEdge, Part.Face)
<class 'UnboundLocalError'>: local variable 'BendEdge' referenced before assignment
```

[image]

Steps to reproduce : 

- create a rectangle in a sketch 
- with the "Make Base Wall" tool, pad the sketch
- select the 4 sides of the plate, and click the "Make Wall" button to add bent sides
- select two continuous edges of the plate to use them with the "Add Corner Relief" tool, the error appears here.

Or use this file :
[SheetMetalIssue.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/10944800/SheetMetalIssue.FCStd.zip)

System info :
- FreeCAD v0.20.2, on GNU/Linux host (Flatpak)
- SheetMetal WB v0.2.61

Thanks :)

nm2107, 2025-01-17:
Closing as not reproducible anymore.
SheetmetalWB version 0.7.3
FreeCAD version 1.0.0

closed 2025-01-17 by nm2107

shaise, 2025-01-17:
Sorry about missing this bug. Glad it works now.