shaise/FreeCAD_SheetMetal #266: Bug: Sequence of folds doesn't work correctly

geogeim, 2023-03-19:
If a do multiple folds, one after another the result is... weird

[image]
[Base.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/11010843/Base.zip)

luzpaz, 2023-03-19:
Full [About](https://wiki.freecad.org/About) info please.

geogeim, 2023-03-19:
> Full [About](https://wiki.freecad.org/About) info please.

[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: Romanian/Romania (ro_RO) ]
Installed mods: 
  * sheetmetal 0.2.61
[/code]