shaise/FreeCAD_SheetMetal #269: Unable to unfold

olivierCHAM, 2023-03-24:
The attached design can not be unfold with sheet metal 0.2.61

The attached file includes an unfold done with sheet metal 0.2.59

OS: Mageia 7 (WindowMaker/WindowMaker)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * fasteners 0.4.54
  * A2plus 0.4.60i
  * workfeature-macro
  * sheetmetal 0.2.61
[C520-210-pliage.FCStd.gz](https://github.com/shaise/FreeCAD_SheetMetal/files/11060229/C520-210-pliage.FCStd.gz)

Syres916, 2023-03-24:
As I mentioned in the forum, please update to 0.2.62 and retry, thanks @shaise for merging so quickly.

olivierCHAM, 2023-03-24:
I have updated to 0.2.62 and the unfold is OK
Thank you all for your response

GS90, 2023-03-25:
Version: 0.2.62
Unstable operation "Unfold"...

[image]

error:
**Unfold is failing.
Please try to select a different face to unfold your object.**
```
Engineering mode is not enabled.
Manual K-factor is being used: 0.50 (ansi)
exception at line 3346("'FeaturePython' object has no attribute 'Refine'",)
("'FeaturePython' object has no attribute 'Refine'",)
Traceback (most recent call last):
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 3346, in accept
    s, foldComp, norm, thename, err_cd, fSel, obN  = getUnfold(k_factor_lookup)
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 2626, in getUnfold
    TheTree.Bend_analysis(f_number, None) # traverses the shape and builds the tree-structure
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 1354, in Bend_analysis
    parent_node = self.make_new_face_node(face_idx, parent_node, parent_edge, wires_edge_lists)
  File "/home/***/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 1054, in make_new_face_node
    if FreeCADGui.Selection.getSelection()[0].Refine is True:
AttributeError: 'FeaturePython' object has no attribute 'Refine'
```

```
OS: Ubuntu 22.04.2 LTS (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.12.6
  * Plot 2022.4.17
  * sheetmetal 0.2.62
  * CfdOF 1.21.7
  * fasteners 0.4.55
```

[detail.tar.gz](https://github.com/shaise/FreeCAD_SheetMetal/files/11068414/detail.tar.gz)

Syres916, 2023-03-25:
Thanks for the report @GS90 , spent so much time testing complex models that I hadn't properly taken into account models without the existence of the Refine attribute. Hopefully I've covered off as many as possible, obviously there are occasions when it will still fail but the Warning message has been enhanced to advise changing Refine to false if it's available. See https://github.com/shaise/FreeCAD_SheetMetal/pull/270

Syres916, 2023-03-29:
@GS90 and @olivierCHAM please update to version 0.2.63 and specifically for @olivierCHAM if OK please close this issue, thanks.

olivierCHAM, 2023-03-29:
Thank you all

closed 2023-03-29 by olivierCHAM