shaise/FreeCAD_SheetMetal #272: Problem with unfold and sketch on sheet metal functions

Pendelf, 2023-03-28:
https://www.youtube.com/playlist?list=PLjxoqfJedrUZft-Fot2py-xC6SsSv7PSx

luzpaz, 2023-03-28:
Missing full [About](https://wiki.freecad.org/About) info.

Pendelf, 2023-03-29:
https://youtu.be/ANFXfNuRbB8

[code]
OS: Manjaro Linux (KDE/plasmawayland)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * sheetmetal 0.2.62
  * A2plus 0.4.60k
  * BIM 2021.12.0
  * Curves 0.6.8
  * dodo 1.0.0
  * ExplodedAssembly
  * fasteners 0.4.55
  * LCInterlocking
[/code]

Syres916, 2023-03-29:
@Pendelf please upgrade to the Wb to version 0.2.63 so we're both using the same version during my testing, some unrelated bugs were fixed.

I'll do some investigation using 0.2.63 but this specific procedure isn't something that I've had to do in real life work so it might take some time.

shaise, 2023-03-29:
the unfold property is not useful in this case. if you want to unfold the model, just select any flat surface and click the unfold toolbar button.
[image]

Pendelf, 2023-03-30:
yeah, that’s what I was doing. I use the unfold property when make a sketch on a face. When a model is finished for unfold i use unfold toolbar button

shaise, 2023-03-30:
The unfold property only works if there are only fold actions for the model. If you add other actions such as cutting holes, the unfold property will not work.

closed 2025-01-06 by shaise

Pendelf, 2025-01-12:
Issue is closed, but it's not clear if it works or not? Maybe I'm doing something wrong?

https://github.com/user-attachments/assets/c8fea218-c121-480c-a2a8-7c2618755d97

shaise, 2025-01-12:
@Pendelf , using the unfold property is not the correct way to work with the sheet metal WB. It will not work if you add features such as holes.
There are two ways to work with that:
1. start with a flat plane, create all the holes you need, then add the folds using bend on sketch. To unfold use the Unfold command icon.
2. create sheet metal model, add holes using sketches using the Extruded Cutout command (works better then Sketch on Sheet Metal) then, again, use the Unfold command to unfold. Do not use the unfold property.

See video:
https://youtu.be/oc_epUv382A

Pendelf, 2025-01-13:
@shaise, In the second case, the rounding on the fold is not correct

[image]

shaise, 2025-01-13:
@Pendelf ,
Of course! because it was originally cut in diagonal. This is used for example if you want to match the sheetmetal to an extruded profile:
[image]

Otherwise you either need to use option 1, or make 2 sketches each flat on the base or the flange, and make 2 separate cuts.

Pendelf, 2025-01-13:
@shaise Ok, I get it, anyway thanks for the work done and for developing the workbench. I'll attach another video to clarify what I'm talking about. First we create the bends we want, then we temporarily straighten them, then we make the cut we want and bend them back again

https://github.com/user-attachments/assets/50cb6139-9e41-4e8a-b9ef-280f4cdf02b0

shaise, 2025-01-13:
@Pendelf , Yes, I understood your problem. Its just that this workbench unfortunately does not support that kind of workflow. It may look simple but its mathematically complicated to implement.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]