shaise/FreeCAD_SheetMetal #276: Changes in previous steps, doesn't work properly.

jarek556, 2023-04-16:
One example: change of bend angle from 90 to -90, creates artificial bend in random place:
[image]
[image]
If it is last step, it works fine.

Full project here:
[grill-box6.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/11243308/grill-box6.zip)

luzpaz, 2023-04-16:
Missing [full About info](https://wiki.freecad..org/About).

jarek556, 2023-04-16:
[code]
OS: Debian GNU/Linux 11 (bullseye) (GNOME/default)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * sheetmetal 0.2.63
[/code]

shaise, 2023-08-02:
This a known freecad issue. changing early steps might change the naming of features causing later steps to be applied on wrong faces. One way to overcome it is to change the early feature then one by one walk through the other steps and reassign the correct selected face to the bend.