shaise/FreeCAD_SheetMetal #279: why unfold  failed

arthur9188, 2023-06-26:
Engineering mode is not enabled.
16:55:41  Manual K-factor is being used: 0.40 (ansi)
16:55:41  exception at line 3367('Removing splitter failed',)
16:55:41  ('Removing splitter failed',)
16:55:41  Traceback (most recent call last):
16:55:41    File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalUnfolder.py", line 3367, in accept
    s, foldComp, norm, thename, err_cd, fSel, obN  = getUnfold(k_factor_lookup)
16:55:41    File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalUnfolder.py", line 2645, in getUnfold
    TheTree = SheetTree(o.Object.Shape, f_number, k_factor_lookup) # initializes the tree-structure
16:55:41    File "C:\Users\Administrator\AppData\Roaming\FreeCAD\Mod\sheetmetal\.\SheetMetalUnfolder.py", line 452, in __init__
    withoutSplitter = self.__Shape.removeSplitter()
16:55:41  Part.OCCError: Removing splitter failed
16:55:41

luzpaz, 2023-06-26:
* Missing [About](https://wiki.freecad.org/About) info
* Missing example file (zip and uploaded to github)
* Missing step-by-step instructions on how to recreate the issue

arthur9188, 2023-06-27:
[https://github.com/arthur9188/freecad-danguidiao](url)
 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[code]
OS: Windows 10 build 17763
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Chinese/China (zh_CN)
Installed mods: 
  * A2plus 0.4.60n
  * Assembly4 0.50.2
  * sheetmetal 0.2.63
[/code]

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise