shaise/FreeCAD_SheetMetal #284: Extra lines on bends in unfold sketch

Trakyan, 2023-11-02:
I've noticed an issue where bends represented by splines often have multiple duplicate lines for one bend in the unfold sketch. This can cause some issues for CAM software as it generates a new operation for each of the duplicates, and it breaks up the profile in strange ways. This only seems to occur with bends being represented as splines. Manually deleting the duplicates in unfold sketch generates a usable DXF file. Also exporting a face of the unfold solid with a subshape binder works as well, but loses the bend lines.

Here is an example
[image]
And zoomed in you can see the duplicate lines side by side
[image]

I assumed this was from both faces of the sheetmetal model being overlaid onto each other, and the minor differences between, but in some cases I've seen up to 4 duplicate lines for one bend (or at least thats how many click+del it takes to make all the lines there disappear, it's definately more than two).

Running the sheetmetal workbench 0.3.1 both on the LS3 tip release and 0.21 mainline freeCAD.
System info:
Operating System: Debian GNU/Linux 12
KDE Plasma Version: 5.27.5
KDE Frameworks Version: 5.103.0
Qt Version: 5.15.8
Kernel Version: 6.1.0-12-amd64 (64-bit)
Graphics Platform: X11
Processors: 4 × Intel® Core™ i5-6200U CPU @ 2.30GHz
Memory: 15.5 GiB of RAM
Graphics Processor: Mesa Intel® HD Graphics 520
System Version: ThinkPad X270 W10DG

luzpaz, 2023-11-02:
Full FreeCAD [About](https://wiki.freecad.org/About) info please.

Trakyan, 2023-11-02:
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33675 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.0)
Hash: 41b058e2087de60dc8fef3d3e68c7d0129e13abf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * sheetmetal 0.3.1
[/code]

I think this is what you mean? 
I made a little example file to demonstrate the issue
[Sheetmetal Example.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13244592/Sheetmetal.Example.zip)

This one you need to zoom in quite a bit in certain spots of the spline to see the two lines, but if you look at the sketch elements there's clearly a few too many splines in there.

luzpaz, 2023-11-02:
Can you also test in the development version ?  https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds
(make sure you back up file first)

luzpaz, 2023-11-02:
@shaise Not sure. Is the issue here due to  Sheetmetal wb issue ?

Trakyan, 2023-11-03:
Tried it on the weekly release of freecad mainline
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34836 (Git) AppImage
Build type: Release
Branch: main
Hash: 862eef57eb4aa042f87d9f19dc768a017b17dae6
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * sheetmetal 0.3.1
[/code]

Recreated a test file from scratch just so everythings a clean slate, the duplicate splines still exist.
[Example FC Weekly.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13250362/Example.FC.Weekly.zip)

In this particular example, when I feed it into the CAM software the bend at the spline at the top of the DXF works fine. Even though it has 3 duplicate splines layered on top of each other, they seem to be identical enough that they get ignored. No matter how much I zoom in on the sketch in FC I only see one line.
[image]

The bottom spline however when I zoom in a bit you can see two splines that are very close but not the same, and the CAM software picks up on this and creates a seperate cutting operation for them.
[image]

From what I can tell this only occurs with bends that are represented by splines. The issue is also slightly random. When I create a bend from two edges which are based on mirrored features and should be identical, often one of them will experience this issue and the other will not.

shaise, 2023-11-04:
When using the "Make bend" tool, both walls should be the same length otherwise there are weird edges:
[image]

Weird edges will cause weirs unfolds.
if one wall is shorter then the other, use the simple "Make Wall" tool, then set the needed gap and bend angle.

Trakyan, 2023-11-04:
> When using the "Make bend" tool, both walls should be the same length otherwise there are weird edges: [image]
> 
> Weird edges will cause weirs unfolds. if one wall is shorter then the other, use the simple "Make Wall" tool, then set the needed gap and bend angle.

Even if both walls are the same length, if one has an angled side the bend is still represented by a spline, and the issue can still occur. Even with a really basic 
The issue isn't that the unfold is failing, or producing "weird" edges, in those cases I rework the model to make the bends look normal/right. The unfold is working perfectly and the edges/bends look perfect and as I would expect them to, the only issue is that there are duplicates of the splines in the unfold sketch layered on top of each other. Deleting the duplicates and leaving one results in a perfectly usable DXF, no weird edges no nothing. 

"make wall" is not an appropriate alternative for most of my use cases, as the walls/faces I need are irregular. I also don't always know the angle of the bend, I usually use a plane by 3 points or something similar map mode to put the sketch for the face in position, and the angle is what it is. 

Even "normal" bends suffer from this issue. I tried to make my corner reliefs all so that the bends are as "neat" and basic as possible, no blending between mismatched walls. The 3D model is shown here:
[image]
And the unfold solid is shown here
[image]
But that corner still has the duplicate splines issue.

It's also sometimes unavoidable to have some "weird" bends, like if one face has an angle to one side, like shown here:
[image]
The bend still forms nicely
[image]
Unfold solid looks good
[image]
Unfold sketch also looks good on the surface
[image]
But checking the sketch elements reveals duplicate splines here, and in this particular case if you zoom in a bit you'll see they aren't entirely identical (sometimes the duplicate splines are identical, and don't cause issues)
[image]

shaise, 2023-11-05:
What do you mean by "angled side"? Here I took your example and made both walls the same length. When I applied the "Make Bend" tool, the result was a clean bend:
[image]

With clean bends you should get a clean unfold:
[image]

Trakyan, 2023-11-05:
In the third photo you can see what I mean by angled, where the sides of the face are not perpendicular to the bend line. It seems you got it to unfold cleanly, but in many other cases it doesn't. Here's another example (sorry, I thought I attached it to the last post)
[same Length example.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13258770/same.Length.example.zip)

Clean bends dont always give clean unfold sketches, the corner I showed in my previous post despite making a corner relief to make the bends as simple and clean as possible, still has a spline in the unfold sketch between where you can see the two lines on the unfold solid.

shaise, 2023-11-05:
Again, for the sheetmetal wb this is considered an invalid geometry:
[image]

For the "Make Bend" to work correctly the corner should be flat and square:
[image]

I'm sorry but these are the limitation of SheetMetal.

You can also try "Fold a Wall" using a plate and a sketch, (attached) but I guess this is not what you are looking for. 
[image]

[Sheetmetal fold on sketch.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13259840/Sheetmetal.fold.on.sketch.zip)

luzpaz, 2023-11-08:
Closing ticket.

shaise, 2023-11-11:
Reopening this issue as there is a reasonable workaround for that, suggested by @easyw .
@Trakyan , please test again and let me know if it is good enough for you,

Trakyan, 2023-11-11:
This seems to have fixed the issue. Greatly appreciate it and all the work done on the sheet metal workbench.
A slight side question. I sometimes get 3 sketches generated (unfold sketch, unfold sketch outline, and unfold sketch bends) and sometimes just 2 (unfold sketch, and unfold sketch bends). Unfold sketch and unfold sketch outline always seem to be identical so I'm just curious if there is supposed to be any difference between them,

shaise, 2023-11-11:
Yes, If you have holes in the part, the outline will be without the holes and the general sketch will be with.

PR #433 (2025-01-06): add new unfolder backend
This PR is to make sure we don't forget to add the new dependency in `package.xml` and close relevant issues. @shaise will plug in the backend functions to the GUI as needed.

This new unfolder:
- Uses the `networkx` python module for analyzing the sheet faces
- Should be indifferent to refined vs. unrefined shapes. Seam lines don't matter in most cases
- Is **5-50x faster than the old unfolder**  

closes #408
closes #361 
closes #345
closes #284 
closes #279 
closes #278 
closes #274 
closes #272 
closes #254 
closes #235 
closes #225 
closes #203 
closes #25 
closes [... cut]
closed 2025-01-06 by shaise