shaise/FreeCAD_SheetMetal #286: Changing the workbench's license

prokoudine, 2023-12-05:
The Workbench Creation guide [recommends](https://wiki.freecad.org/Workbench_creation) licensing addons under the terms of LGPL2.1 — the same license that FreeCAD source code is available under. The recommendation’s rationale is that it should be possible to ship a FreeCAD distribution, either upstream, or a custom one, with 3rd-party addons that are considered useful.  While a workbench developer is free to choose an alternative license, doing so may impact the long-term viability and use of the workbench and should be considered carefully.

Ondsel would like to integrate the Sheet Metal workbench into its distribution of FreeCAD, because the company’s market study suggests that, just like iwth the Fasteners WB, this feature is expected to be available in 3D CAD programs by default. We want to provide our end-users the best possible experience. There’s a possibility that this workbench then will go back to become part of upstream FreeCAD. However neither the upstream version of FreeCAD, nor Ondsel’s version of FreeCAD can distribute this workbench as long as it’s licensed under the terms of GPL3.

We are asking for a conversation about relicensing this workbench under LGPL2+ or another compatible license to allow wider distribution.

shaise, 2023-12-05:
I'm all for re-licensing to LGPL2.1, this is also compatible with @ulrich1a screw_maker license which the original WB was based on.

sliptonic, 2023-12-06:
As a contributor, I'm fine with my contributions being relicensed.

shaise, 2023-12-07:
waiting for response of:
- [X] @shaise 
- [ ] @jaisekjames
- [ ] @luzpaz
- [ ] @Syres916
- [ ] @easyw
- [ ] @chappatted
- [ ] @realthunder
- [X] @sliptonic
- [ ] @riqueenz
- [ ] @Roy-043
- [ ] @looooo
- [ ] @hasecilu
- [ ] @alexneufeld
- [ ] @mangelozzi
- [ ] @rmu75
- [ ] @chennes
- [ ] @s-light
- [ ] @kaktusus
- [ ] @ufocia
- [ ] @ASzc
- [ ] @Jantek
- [ ] @bgamari
- [ ] @hl037
- [ ] @ceremcem
- [ ] @thesyndicated
- [ ] @mumme74

rmu75, 2023-12-07:
My contribution isn't really noteworthy, I'm ok with relicensing.

riqueenz, 2023-12-07:
No objection to the license change!

hl037, 2023-12-07:
No objection on my side neither

hasecilu, 2023-12-07:
I agree with the change!

s-light, 2023-12-07:
i am happy with the change!

Jantek, 2023-12-07:
> [Verfolgungsbild][https://github.com/notifications/beacon/AAU5EKXU5QLRKY5UWCAF4FTYIIMNJA5CNFSM6AAAAABAIGRKK2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTTOA64BA.gif]

I am ok with the change

kaktusus, 2023-12-07:
As they say in my country I am for and even against
This is an ironic admission that the previous speaker is right.
:laughing:

chennes, 2023-12-08:
I agree with the change.

ufocia, 2023-12-08:
I'm OK with LGPL2.1.

mangelozzi, 2023-12-08:
I agree with the change. Totally makes sense to integrate it

thesyndicated, 2023-12-08:
I think I only fixed a typo here, but I'm totally good with the license
change.

realthunder, 2023-12-08:
No objection! Congratulations!

JMG1, 2023-12-08:
No objections from my side!

luzpaz, 2023-12-08:
No onjection

easyw, 2023-12-15:
go for it!
sorry being late, but I don't watch very often my GH notifications

mumme74, 2023-12-21:
No objection, sorry for responding late.
Change as you se fit as long as its an open source license.

shaise, 2023-12-21:
@ceremcem , @Syres916 

Your responses?

Syres916, 2023-12-21:
No objection

prokoudine, 2023-12-21:
@ASzc What's your take on this?

ASzc, 2024-01-07:
I'm ok with re-licensing to LGPL2.1

prokoudine, 2024-01-07:
Only up to you now, @bgamari and @ceremcem!

ceremcem, 2024-01-11:
Hi. Sorry for the delay. I'm okay with the change you propose.

bgamari, 2024-01-15:
I am fine with relicensing.

shaise, 2024-01-15:
Great!! All in!
@prokoudine , what is the next step?

shaise, 2024-01-15:
if you can do it, that would be great!

shaise, 2024-03-31:
Close as accomplished.

closed 2024-03-31 by shaise