shaise/FreeCAD_SheetMetal #289: Unfold : export sketch dxf  ===>  does not create a dxf file

free777cat, 2024-01-03:
"Export sketch" does not create a dxf file.
The resulting SVG file is marked as DXF.
[image]

A folder with an old version that works.
[sheetmetal_work_dxf.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13818456/sheetmetal_work_dxf.zip)

shaise, 2024-01-04:
You are correct! 
Thanks for the catch. 
It should be ok now, please update and test.

free777cat, 2024-01-04:
[image]
[image]

shaise, 2024-01-04:
> Question number 2
> Is it possible to make the selection options so that there is only the filename? 

No because there are times that you have several unfold items in the same file.

free777cat, 2024-01-04:
https://github.com/shaise/FreeCAD_SheetMetal/issues/289#issuecomment-1877066085

free777cat, 2024-01-04:
Such "extended" file names are very disturbing for further work.

free777cat, 2024-01-05:
Maybe you will make it possible to choose a "mask" of file names? Of course I can manually fix the code, but having a "switch" would be a better solution.

[image]

closed 2024-03-27 by free777cat