shaise/FreeCAD_SheetMetal #290: TypeError in generateSketch

ksigurdur, 2024-01-09:
Ubuntu 20.04.3
FreeCAD 0.20.1
SheetMetal WB  0.3.14

Unfold was successful but only the "unfold" sketch was being generated. The additional sketches (outline, internal, and bends) were failing to generate.

`12:53:33  Traceback (most recent call last):
  File "/home/[...]/.local/share/FreeCAD/Mod/sheetmetal/./UnfoldGUI.py", line 213, in accept
    raise e
  File "/home/[...]/.local/share/FreeCAD/Mod/sheetmetal/./UnfoldGUI.py", line 179, in accept
    result = smu.processUnfold(
  File "/home/[...]/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 3169, in processUnfold
    unfold_sketch = generateSketch(edges, "Unfold_Sketch", sketchColor)
  File "/home/[...]/.local/share/FreeCAD/Mod/sheetmetal/./SheetMetalUnfolder.py", line 3255, in generateSketch
    sk.ViewObject.LineColor = rgb_color
<class 'TypeError'>: {'sclassname': 'N4Base9TypeErrorE', 'sErrMsg': 'Type in tuple must be float', 'sfile': '', 'iline': 0, 'sfunction': '', 'swhat': 'Type in tuple must be float', 'btranslatable': False, 'breported': False}`

Modifying line 3254 in SheetMetalUnfolder.py from

`rgb_color = tuple(int(color[i : i + 2], 16) for i in (1, 3, 5))`

to

`rgb_color = tuple(float(int(color[i : i + 2], 16)) for i in (1, 3, 5))`

appears to fix the error.  All four sketches get generated.

shaise, 2024-01-10:
Hi @ksigurdur 
this does not happen in the latest stable version of freecad (21.1), but I will take a look and if it is compatible, I will fix it

shaise, 2024-01-10:
Done!

Thank you for the fix!

ksigurdur, 2024-01-10:
Thanks @shaise, I appreciate the prompt response. Issue fixed and closed.

closed 2024-01-10 by ksigurdur