shaise/FreeCAD_SheetMetal #295: [feature] Fold non parallel sheet

Anthony-Gaudino, 2024-01-17:
It seems that currently SheetMetal can only create bends on sheets with parallel surfaces.

Trying to create a bend where the bottom is not parallel results in a wrong bend:
[image]
[bad_bend.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13968345/bad_bend.FCStd.zip)

If both top and bottom are not parallel it gives an error:
```
SheetMetalFoldCmd.py", line 266, in execute
    s = smFold(bendR = fp.radius.Value, bendA = fp.angle.Value, flipped = fp.invert, unfold = fp.unfold, kfactor = fp.kfactor, bendlinesketch = fp.BendLine,
  SheetMetalFoldCmd.py", line 122, in smFold
    toolFaces = tool.extrude(normal * -thk)
<class 'Part.OCCError'>: BRepSweep_Translation::Constructor
20:19:20  Recompute failed!
```

[broken_bend.FCStd.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/13968417/broken_bend.FCStd.zip)

------

Having the possibility of bending non parallel surfaces would be very useful in some instances.

shaise, 2024-01-17:
Yes. This a sheet metal limitation. It was meant to be used with flat sheets of metals.

luzpaz, 2025-12-16:
> This a sheet metal limitation. It was meant to be used with flat sheets of metals.

@shaise are there plans to add this functionality eventually to SheetMetal WB ?

shaise, 2025-12-17:
@luzpaz , I do not think this feature should be implemented. Its not in the scope of sheetmetal.