shaise/FreeCAD_SheetMetal #296: [feature] refold unfolded sheet

Anthony-Gaudino, 2024-01-17:
It would be great if we could refold an unfolded sheet metal.
This way users could unfold, make changes, and then refold.

shaise, 2024-01-17:
I agree this can be a nice feature.

maxwxyz, 2024-03-31:
To add:
Maybe the unfold can become a standard feature in the same tree, not a separate object.Then you could unfold, add other features (maybe a pocket) and refold (all or single bends) in the following feature. So it will stay parametric.

shaise, 2024-03-31:
Yes, this is a very much desired feature, but quite hard to implement without refactoring the base algorithms sheetmetal uses. Hopefully we get there as well

EstebanFPV, 2024-11-27:
this could potentially be somewhat solved by being able to create sheetmetal parts out of the faces of solid objects, though i agree it would be pretty sweet