shaise/FreeCAD_SheetMetal #297: [feature] fold wires (cylindrical surfaces)

Anthony-Gaudino, 2024-01-17:
It would be very nice if we could use SheetMetal to fold and unfold wires (cylindrical surfaces).

[image]

shaise, 2024-01-17:
In case of wires its much more easy to just sketch the wire then loft it with a circle.

Anthony-Gaudino, 2024-01-17:
Yes, it's easier, but I believe it would be useful.

In my case for example, I have a wire that has bends and some other complex things that would be much easier to model if I start from a flat wire.

So I could model the complex stuff on the flat wire and then bend it using sheet metal.
Also I would avoid creating "3D sketches".