shaise/FreeCAD_SheetMetal #303: Extruded cutouts

pierreporte, 2024-03-01:
`SMSketchOnSheet` can be used to make cutouts on sheet metal parts but it is very simple: shapes are drawn on the flat sheet. Users can’t easily remove material where an object is going to pass through the cutout. Using `PartDesign_Pocket` to do so can work but the cross section isn’t right. It needs a new feature that cut the part in a correct way for manufacturing. The picture belows illustrates the feature request.

[image]

The sketch represent the object that the part need to accomodate for. It isn’t necessarily on a face nor parallel to one. Like `PartDesign_Pocket`, there should be an option like “up to first”, “up to surface”, and “up to last”. A and B are where the sheet is always getting cut. B and E are where the sheet is getting cut if the user selects the option to cut at the intersection of the cutout and the reference plane of the sheet metal. C and F are where the sheet is getting cut if the user selects the option to remove all needed material. Of course it should work if bends are in the projected space of the sketch.

sheetmetalman, 2024-09-29:
Forum topic: [Sheet metal - normal cut (or extruded cutout)](https://forum.freecad.org/viewtopic.php?p=784731#p784731)

pierreporte, 2024-09-29:
@shaise Do you think that the method proposed in the linked macro good enough to be part of the workbench?

shaise, 2024-09-29:
I need to investigate it, but I'm quite busy these days.

jaisekjames, 2024-10-07:
In SheetMetalExtendCmd.py,  code starting from line 274 will be used for similar purpose. Same functions can reused for above purpose.

sheetmetalman, 2024-10-08:
@jaisekjames Awesome!

I read little bit of this code you mention. If I understand correctly, it can detect the correct surfaces of the sheet metal (not the thickness surfaces) and cut. I don't understood how it can detect that. If someone can help me to undertand this detection part, I can make progress in my macro ([see the this forum topic](https://forum.freecad.org/viewtopic.php?t=91033)). And close this issue, if I successful.

pierreporte, 2024-10-08:
The issue can be closed only if the macro is part of the workbench though. Do you think you can integrate it in the code?

sheetmetalman, 2024-10-08:
Of course. I don't know if I can, 'cause I never made it before. But, If I can complete the macro, it's a few steps near the be successful code. After this, I have to discover how to made the macro part of the workbench. Every help is welcome.

shaise, 2024-10-09:
Please let me know when you macro is complete

sheetmetalman, 2024-10-10:
Ok!

Now, I'm going to study the SheetMetalExtendCmd.py code (mainly around the line 274) to learn how to code the offsets part.

sheetmetalman, 2024-11-05:
import FreeCAD as App
import FreeCADGui as Gui
import Part

# Ensure there is a selection
selection = Gui.Selection.getSelection()
if len(selection) == 0:
    raise Exception("No face selected. Please select a face.")

# Get selected sketch
cutSketch = Gui.Selection.Filter("SELECT Sketcher::SketchObject")
cutSketch.match()
cutSketch = cutSketch.result()[0][0].Object

# Get the selected face
selected_object = selection[0]
if not selected_object.Shape or len(selected_object.Shape.Faces) == 0:
    raise Exception("No valid shape found on the selected object.")

# Get the selected face directly from SubObjects
selection_ex = Gui.Selection.getSelectionEx()[0]
selected_face = selection_ex.SubObjects[0]

# Get the normal of the selected face
normal_vector = selected_face.normalAt(0, 0)

# Get all faces of the object
faces = selected_object.Shape.Faces

# Initialize variable to store the minimum distance
min_distance = float('inf')

# Step 1: Find the thickness of the sheet metal part
for face in faces:
    if face is not selected_face:
        if normal_vector.isEqual(face.normalAt(0, 0).multiply(-1), 1e-6):
            distance_info = selected_face.distToShape(face)
            distance = distance_info[0]
            if distance < min_distance:
                min_distance = distance

if min_distance == float('inf'):
    raise Exception("No opposite face found to calculate thickness.")

thickness = min_distance
print(f"Thickness of the sheet metal part: {thickness:.4f} units")

# Step 2: Find all pairs of parallel faces with a distance equal to the thickness
parallel_faces = []

for i, face1 in enumerate(faces):
    for j, face2 in enumerate(faces):
        if i >= j:
            continue

        if face1.normalAt(0, 0).isEqual(face2.normalAt(0, 0).multiply(-1), 1e-6):
            distance_info = face1.distToShape(face2)
            distance = distance_info[0]

            if abs(distance - thickness) < 1e-6:
                parallel_faces.append(face1)
                parallel_faces.append(face2)

if parallel_faces:
    print(f"Found {len(parallel_faces) // 2} pairs of parallel faces with thickness distance:")
    for idx in range(0, len(parallel_faces), 2):
        f1 = parallel_faces[idx]
        f2 = parallel_faces[idx + 1]
        print(f"Pair {idx // 2 + 1}: Face1 at {f1.CenterOfMass} and Face2 at {f2.CenterOfMass}")

    shell = Part.Shell(parallel_faces)
else:
    print("No pairs of parallel faces with the specified thickness distance were found.")

myFace = Part.Face(cutSketch.Shape)

cutLength = 1000
cutHalfLength = cutLength / 2

normalLength = myFace.Faces[0].normalAt(0, 0) * (cutHalfLength * (-1))
normalLengthRev = myFace.Faces[0].normalAt(0, 0) * cutHalfLength

myExtrusion1 = myFace.extrude(normalLength)
myExtrusion2 = myFace.extrude(normalLengthRev)

myUnion = Part.Solid.fuse(myExtrusion1, myExtrusion2)
myUnion = myUnion.removeSplitter()

myCommon = myUnion.common(shell)

# Step 3: Create separate objects for each group of connected surfaces
def find_connected_faces(shape):
    faces = shape.Faces
    visited = set()
    components = []

    def is_connected(face1, face2):
        for edge1 in face1.Edges:
            for edge2 in face2.Edges:
                if edge1.isSame(edge2):
                    return True
        return False

    def dfs(face, component):
        visited.add(face)
        component.append(face)
        for next_face in faces:
            if next_face not in visited and is_connected(face, next_face):
                dfs(next_face, component)

    for face in faces:
        if face not in visited:
            component = []
            dfs(face, component)
            components.append(component)

    return components

connected_components = find_connected_faces(myCommon)

offset_shapes = []  # List to hold all the offset shapes

for index, component in enumerate(connected_components):
    if component:
        # Create a shell from the connected faces
        component_shell = Part.Shell(component)

        # Check if the shell is valid
        if component_shell.isValid():
            # Create a filled offset shape with reversed thickness
            try:
                offset_shape = component_shell.makeOffsetShape(-thickness, 0, fill=True)

                # Ensure the offset shape is valid
                if offset_shape.isValid():
                    offset_shapes.append(offset_shape)  # Add the valid offset shape to the list
                    print(f"Offset shape for component {index + 1} created successfully.")
                else:
                    print(f"Offset shape for component {index + 1} is not valid.")
            except Exception as e:
                print(f"Error creating offset shape for component {index + 1}: {e}")
        else:
            print(f"Component {index + 1} is not a valid shell.")

# Now, union all the offset shapes
if offset_shapes:
    # Create the first solid from the first offset shape
    combined_offset = Part.Solid(offset_shapes[0])
    for shape in offset_shapes[1:]:
        combined_offset = combined_offset.fuse(shape)  # Union with each subsequent offset shape

    # Cut the combined offset from the selected object
    cut_result = selected_object.Shape.cut(combined_offset)
    new_object = App.ActiveDocument.addObject("Part::Feature", "CutResult")
    new_object.Shape = cut_result
    print("Created cut object: CutResult")
else:
    print("No valid offset shapes were created.")

App.ActiveDocument.recompute()

sheetmetalman, 2024-11-09:
Videos of the first parametric version. Still in testing phase.

https://github.com/user-attachments/assets/e3fdc16d-ffb7-4469-8f9b-924cee862c81

https://github.com/user-attachments/assets/1d3191b4-cfa7-4bf7-857e-d392fa7ec5a5

shaise, 2024-11-09:
Great work!

pierreporte, 2024-11-09:
I don’t fully understand how it works by looking at the video. It looks like the extrusion is not normal to the sketch plane.

sheetmetalman, 2024-11-09:
The extrusion is normal to sketch plane.

These videos (from before the parametric version - but works in the same way) shows it better:

https://github.com/user-attachments/assets/5de02f60-4ebf-4061-9cdd-ffe1ee784685

https://github.com/user-attachments/assets/b897cec5-11d5-479f-8254-5554f4a238b0

shaise, 2024-11-09:
@pierreporte ,

Basically this feature allows you to cut an extruded profile that is diagonal to the sheet part, but the sheet part retains its perpendicular edges.
This keeps the validity of the sheet metal part.
[image]

pierreporte, 2024-11-09:
@sheetmetalman Thanks for the new video. I can see better how the sketch is oriented. The result is perfectly fine and what I wanted when opening the issue.

@shaise We actually agree. It’s all clear now.

Do you think it should replace `SheetMetal_SketchOnSheet`?

shaise, 2024-11-09:
@pierreporte ,
Good question. This can be a good idea, however this will not be backward compatible with old files.

pierreporte, 2024-11-09:
If the new feature can replicate how the current one works, it is possible to convert on the fly. In any case, the old feature can stay in the code while hidden to users.

sheetmetalman, 2024-11-09:
I think the SketchOnSheet tool must be keep. It's a great feature that can work for other purposes, like text (or other geometries) curved along a round surface. As far as I know Solidworks don't have a tool like this for sheet metal parts.

Maybe it can be renamed (at least in the icon name) for something like "Wrap cut".

I'll send a picture where SketchOnSheet tool can be used better than other tool.

[image]

closed 2024-11-11 by pierreporte

shaise, 2024-11-11:
@pierreporte do you mean sketch on the unfolded part then fold it back? Dont think its that simple.

pierreporte, 2024-11-11:
@shaise Yes exactly. But it would need fixing #202 first at the very least.

shaise, 2024-11-11:
Even if you have a parametric unfolded solid, adding cuts to it and then refolding it creates some kind of a cyclic dependency.

pierreporte, 2024-11-11:
Not really because it would be the unfolded part when the feature is applied. Basically the unfolded view would be different depending on the tree tip. Catia works like this: if you want to make a cutout like the older FreeCAD feature, you have to activate the unfolded view, create the sketch, apply the cutout, and then activate the folded view back. But it’s even more complicated I guess.