shaise/FreeCAD_SheetMetal #304: Clever bending direction

pierreporte, 2024-03-01:
When selecting an edge for `SMMakeWall`, the fold is always in the same direction relatively to the reference surface of the part. It could be in the direction closer to the selected edge, while still allowing to invert it

[image]

shaise, 2024-03-02:
This is already the case - just to the opposite direction....

pierreporte, 2024-03-02:
I feel dumb to not have understood it because that works all the time. You can close the issue. Why the behavior is inverted?

shaise, 2024-03-02:
actually as i remembered it was the other way around. I need to ckeck old versions. perhaps its a bug

shaise, 2024-03-04:
Yes, indeed it was the correct way before and was changed sometime at 2022. I will fix it even if it brakes saved models.

shaise, 2024-03-10:
issue fixed now.

pierreporte, 2024-03-10:
Thank you for the fix.

closed 2024-03-10 by pierreporte