shaise/FreeCAD_SheetMetal #305: `SMFormingWall` is too complicated for most cases

pierreporte, 2024-03-02:
`SMFormingWall`  requires the user to make a different solid with the inverted shape they want to be stamped into the part. It allows a lot of freedom and doesn’t limit the user to a few simple shapes. However, most of the time, stamping is actually pretty simple: a flat bottom, a cutout, etc.

The workbench could provide a simplified version of `SMFormingWall` that takes a sketch as an input. The sketch would have been the outer shape of the solid. Potentially, the sketch could have several closed profiles. This new feature would provide some options to make a cutout (instead of selecting the opposite file) and allow to select edges in the sketch to mark that there must be an opening there (to make bridges or louvres). `SMFormingWall` would be used for the most complex cases.

Below are some examples from Solidworks and Catia to show how it could look.

This issue has been thought of with the FreeCAD Design Working Group initiative for the sheet metal worbench conducted by @obelisk79 for @sliptonic. It is posted there because it is out of scope of this work.

--------------

[image]

Catia surface stamp: http://catiadoc.free.fr/online/cfyugstamp_C2/cfyugstampsurface.htm  
[image]

Catia flanged cutout: http://catiadoc.free.fr/online/cfyugstamp_C2/cfyugstampflangedcutout.htm  
[image]

Catia flanged hole: http://catiadoc.free.fr/online/cfyugstamp_C2/cfyugstampflangedhole.htm  
[image]

pierreporte, 2024-11-11:
@shaise What do you think of this?

shaise, 2024-11-11:
Yes, totally agree the forming tool is very complicated. But the idea of designing a solid with the desired punch shape seems to me more intuitive. The only problem is how to simplify the UI to use it.
A good task dialog can be a solution for that. Right now only the bend tool and base shape have usable task panels. I want to add better task panels for the other tools.

pierreporte, 2024-11-11:
Designing a solid is not bad but it should be used only for the most complex cases. Even with a good UI, you’ll have some extra steps that are not necessary.

@FreeCAD/design-working-group can we propose an easy to use UI for this?

shaise, 2024-11-11:
The good thing about using a sketch is that it does not involve using solids from other bodies (which tend to complicate part design, not only for sheetmetal) So I'm definitely in favor of adding this option as well. Coding-wise its quite complicated.

pierreporte, 2025-02-10:
@sheetmetalman What about this?

shaise, 2025-02-10:
@pierreporte , I'm already working on a task panel for that.

pierreporte, 2025-02-10:
@shaise Great!

sheetmetalman, 2025-02-11:
@pierreporte Indeed it will be a great improvement for `SheetMetal_Forming`!
I understand that @shaise is already taking care of this!

As @shaise said and looking at that Catia manual, I think using a sketch could be a good thing for this tool. Together with taskpanel will be asy to use and understand. Better UI/UX!

Perhaps an initial idea could be that the user could select a solid or a sketch for reference.
If the user selects a solid, it would work as it currently does.
But if user select a sketch, the tool will first create an extrusion and then use this extrusion in the same way the tool uses the solid. Of course, that extrusion will not be an object in the model tree, it will only exist in code.