shaise/FreeCAD_SheetMetal #306: `SMFoldWall` should display an error message if the profile is not colinear lines

pierreporte, 2024-03-02:
*This issue has been thought of with the FreeCAD Design Working Group initiative for the sheet metal worbench conducted by @obelisk79 for @sliptonic. It is posted there because it is out of scope of this work.*

`SMFoldWall` needs that the sketch contains only one fold line, continuous or discontinuous. If it is discontinuous, the segments must be colinear. If one of these conditions is not fulfilled, the feature fails. However, in certain cases, it still produces a result but obviously wrong. When there is an error message, it is not helpful. See the pictures below.

`SMFoldWall` should check if the profile is correct and display an error message listing what’s wrong: several profiles, non-colinear fold line, etc. and not produce anything.

------------------------

[image]

[image]

[image]

[image]

[image]

[image]

[image]

shaise, 2024-03-03:
I think this feature should be limited to a single straight line. Otherwise it should report an error

pierreporte, 2024-03-03:
The wiki shows a very legitimate use of a discontinuous line: https://wiki.freecad.org/SheetMetal_AddFoldWall

Maybe it would be better to graphically select in the 3D view the parts that shouldn’t be bent.

shaise, 2024-03-03:
Ohh, Yes, correct. so in this case, checking the validity of such line is not so easy.

pierreporte, 2024-03-03:
I think that you can programmatically get a list of all geometry in a sketch. It may be possible to check the colinearity. Would it be enough to check numerically if all vertex sit on a line?

pierreporte, 2024-03-03:
Apparently, OpenCascade provides a way to verify if a set of points lie on a line: https://dev.opencascade.org/doc/refman/html/class_g_prop___p_equation.html

pierreporte, 2024-09-20:
There can be a workaround to avoid checking colinearity of segments and using only a single line, but it would require FreeCAD sketcher to allow users to select zones of sketches. This has been requested but it it yet to be implemented. With this, Sheet Metal workbench could cut the part in two halves, and those halves would be divided in several closed zones (like a projection of the whole contour, cut by the line). Then users could select the zone they don’t want to be bent.

pierreporte, 2024-11-11:
As written in #307, a solution could be to only accept sketches with one single line, and allowing users to manually select in the 3D view the excluded zones.