shaise/FreeCAD_SheetMetal #307: Making folded walls on planes or surfaces

pierreporte, 2024-03-02:
In sheet metal design, it is often needed to make flanges to follow certain planes or surfaces that were predefined, usually for enclosing or to support existing geometry. A workaround would be to draw walls at the precise dimensions so that `SMMakeWall` or `SMFoldWall` draw it perfectly where it should but it is not a practical workflow.

Other programs allow to use planes or surfaces as references. Usually, they need to cross the base wall. In other word, the base wall need to go further than the supporting geometry. The workbench would greatly benefit from this. It would be a mix between `SMFoldWall` and `SMMakeWall` because the workflow is similar to fold on line but it needs the parameters of the normal flange: leg length, chamfers, miters, etc. (except angle obviously).

[image]

pierreporte, 2024-11-11:
@shaise What do you think of this feature?

It will need something to take into account zones that shouldn’t be bent as illustrated in [AddFoldWall](https://wiki.freecad.org/SheetMetal_AddFoldWall) wiki page. It would be a good idea that the UX is the same for both for these exclusions. I think that the best way would be to allow to select surfaces further than the computed fold line, and it would allow to solve #306 by simply allowing only one continuous line in the sketch.

shaise, 2024-11-11:
This can be a nice feature, I agree

pierreporte, 2024-11-29:
I think this feature could reuse part of `SMFoldWall` by computing a line automatically (but not creating a user-accessible sketch).

It could be a good idea to offer the ability of `SMMakeWall`, that is asking for a leg length, and the ability of `SMFoldWall` that just folds existing geometry. There could be a third option using a sketch that would become the contour of the new flange.

@sheetmetalman if you’re looking for something new, this is one of the last big missing features of the workbench.

sheetmetalman, 2024-11-30:
@pierreporte Hi!

Maybe I got the idea. Maybe not. Tell me if I got the idea right, please.

Maybe it can be done with the `SMMakeWall`. Just need to be added another type of parameter for offset and angle. Instead of only values, can has something more advanced like face selection.

For the angle: the user select a face as a parameter and select if the angle must be parallel or normal to that face.

For the offset: the user select the same face, and the limit will be the intersection of the sheetmetal plane with the plane of the user selected face, on other words the distance between the end of sheet metal to intersecttion.

[image]
[image]

pierreporte, 2024-11-30:
@sheetmetalman That’s pretty much it.

> For the angle: the user select a face as a parameter and select if the angle must be parallel or normal to that face.

It would be better if the user can enter an arbitrary angle, not just 0° or 90°.

> For the offset: the user select the same face, and the limit will be the intersection of the sheetmetal plane with the plane of the user selected face, on other words the distance between the end of sheet metal to intersecttion.

I think that there should be a zero offset by default so that users don’t have to click twice.

sheetmetalman, 2025-01-02:
Hello!

I am implementing these functions. The angle has an option to use a face as reference and the offset has an option to use a face as reference. And I made some changes on the task panel to fit better with these new functions.

It's not done yet.

In my SheetMetalWB fork there are already changes in the `SheetMetalCmd.py` and the `FlangeParameters.ui` files.

Here's two video demonstrating how this is currently working:

https://github.com/user-attachments/assets/5e364176-bc3f-4b63-b610-e01477b6b908

https://github.com/user-attachments/assets/610025e0-4c14-492f-928e-277a2293f9c2

There are some things to improve, some known issues. Maybe someone can help me.

One thing is on the task panel. Maybe will be better has button that keep pressed when the option is on, instead of having a check button and button to select the face.

KNOWN ISSUES:
* I cannot find I way to predict the wall side and if the angle is correctly applied (eg if it was the supplementary angle instead of the correct angle). I tried a lot, and don't find a way. Because of this I implemented the supplementary angle option. If someone know how to achieve that, it will be nice, and we can remove this property. What I needed was get the face of the new created wall, but if has more than one wall created, get only the wall attached to the first baseObject in the list.
* Angle don't updates automatically (but it's 'cause this code works like that, I tried to fix it, but I can't). The user must change de length of the wall for update.
* Negative offset when the reference face is outside the sheet metal will lead to an error.
* When the reference face isn't aligned (eg if its not just parallel rotated) the angle and the offset don't is get correctly.
* Wall on sheet metal hole. The addWall command doesn't work entirely.
* Don't work with datum and origin planes.

shaise, 2025-01-02:
Hi @sheetmetalman ,
Very impressive indeed!
for the first issue, I think the "invert" property can be used instead of adding a new property

pierreporte, 2025-01-02:
Cool!

For datum, maybe @PaddleStroke could help you. He implemented the new system.

I think that the offset system could be more automatic. Usually, when you want to use such feature, you want the flange on the plane or surface.

sheetmetalman, 2025-01-03:
I'm very glad you like it, guys! Thanks!

@shaise . I tried to use the "invert" property, but unfortunately don't work as the needed. The "supplementary" property is the only workaround that actually works. As shown in the video below.
The perfect solution is get the face of the new created wall, but if has more than one wall created, get only the wall attached to the first baseObject in the list. With this the "Supplementary" property can be take off and also dozen of code lines of tests to check the correct side and angle.
If you want to take a look, a few minutes ago I update the `SheetMetalCmd.py` file in my fork.

https://github.com/user-attachments/assets/2cf5c3ef-ed27-47f0-9369-76c1c9b9ef96

@pierreporte Nice! When the @PaddleStroke new feature is available to use on it, I will update the code.
I don't know if I get right what you mean about offset be more automatic. But by if you mean to use the same face reference of the angle, I think the best approach is to have an option to do this (something like a checkbox that copy the angle face reference to offset face reference). Because sometimes the offset reference can be different, or the offset reference is not desirable, or even the offset reference is wished and the angle reference don't. I say this because I also design sheet metal parts professionally.

pierreporte, 2025-01-03:
@sheetmetalman The new datum have been merged for weeks now. You can test them with the weekly version.

What I mean with more automatic is that, most of the time, when you want to use that capability, you need the new flange to be on the selected surface, so with both your new options enabled and with the face or plane selected. Linear or angular offset are more of a special case, so I think that they should be an option of the first case, not the other way around.

BTW, do your code work if the surface intersect the existing geometry, as illustrated in the opening comment of the issue? Also, how do you manage exuding portion of the sheet metal from the bend, like `AddFoldWall` can do? (see https://wiki.freecad.org/SheetMetal_AddFoldWall)

shaise, 2025-01-03:
@sheetmetalman , What i meant about the 'invert' option is not to use it as-is - of course it will not work. I meant that since in your case the original invert has no meaning, you can replace its function with yours, and save an extra parameter.

sheetmetalman, 2025-01-04:
@pierreporte
Nice! I'll take a look at that.
Thanks for explain again to me! But I don't get the point yet. Sorry please! :smiling_face_with_tear: :cry:
About the plane intersect the sheet metal it works as illustrated in the opening comment of the issue. See the video below (where I demonstrate it and that property we talk before about the relative angle - not just 0° or 90°).
About how it manage exuding portion of the sheet metal from the bend: I didn't change or touch the code that creates the walls. So, the AddWall command do the same thing as before. The code I made only the automate the angle calculation to has a starting point parallel to the face reference; and automate the offset calculation to be at the intersection point with the face reference. I also changed the task panel to fit with these new options.

https://github.com/user-attachments/assets/ec0f7200-8923-420d-8f10-ca3a92add666

@shaise Now I get the point. I think the current"invert" property can be useful, I don't think it must be replaced when these options (angle/offset face reference mode) is enabled.
About the face select for the bend, I use it intentionally, 'cause the user **CAN** select it in the AddWall command, when the user select the edge for bend, the code get the side of the bend easily, but when the user select the face it's hard. If the AddWall command restrict the selection for only select edges, well, this will be much easier. In fact, I think, this should be a update in the code. 'cause, when designing sheet metal parts, I never use the thickness face for this type of flange creation (on the software that I used to use), I only use the edges as reference for the wall (in SolidWorks this function is called ["Edge Flanges"](https://help.solidworks.com/2025/English/SolidWorks/sldworks/c_edge_flanges.htm?id=46ff9b80ad6c4a67a0fe696f4a81c957&_gl=1*1ifp37*_up*MQ..*_ga*MjA5ODI3OTM3OS4xNzM1OTUzNTE2*_ga_XQJPQWHZHH*MTczNTk1MzUxNi4xLjEuMTczNTk1MzUzNi4wLjAuMA..#Pg0)).

shaise, 2025-01-04:
@sheetmetalman , regarding what @pierreporte  said, I think that if a user selects a face of another object and then select an edge of a sheetmetal object then in this case you can assume that the user intended for the up-to-face function. No need to select that explicitly.
Point 2: I think I'm going to remove the face select for add wall and only have it as backward compatibility when loading old files. I do not see a case where it is favorable over edge select.

sheetmetalman, 2025-01-04:
Indeed! Removing facing selecting for AddWall command is a very good idea! (Also, the bend side logic when the user selects the thickness face is unpredictable. Sometimes it goes one way, sometimes it goes the other. It was a nightmare trying to predict this. :cold_sweat: I just give up and add that "Supplementary" property. :sweat_smile:)

I see! Selecting the edge and after the face to made that happen. It would be faster to do that.

But that wasn't my idea on the begining. I thought the best approach is just add these options (face reference for angle/offset) in the AddWall command. I don't was thinking about creating a new command for the SheetMetalWB. If it's the case, can't be done using such faster flow, because with AddWall command the user can select multiple edges to create walls.

shaise, 2025-01-05:
Perhaps it is better to have a separate command for it, as the addwall, have too many parameters already. With this new tool we can restrict the selection to a single edge and single face.

sheetmetalman, 2025-01-05:
I think we should consider what is made to implement these functions on other softwares.
For example, in SolidWorks in the "Edgle flange" tool has a face selection mode for angle and for offset (and beyond, has a face/vertex selection to setup the length of the wall, but this is another history).

I think this types of face/vertex/edges selection is basic for this type of things (defining lengths, angles etc. Like the in PartDesign Pad/Pocket tool they have "UpToFace", "ThroughAll" and "UpToShape" etc. modes), it can be very useful and better for automating and parameterizing these functions.

In the image are shown the SolidWorks task panel for face selection for angle and for offset.

[image]

shaise, 2025-01-06:
Ok Sounds good.
I just want the UI to be as simple as possible. perhaps we need to do the same as Pad? Have a new type in selection: "up to face" and then we can show/hide appropriate fields?

sheetmetalman, 2025-01-06:
I too. I want this tool and the whole workbench keep as simple as possible.

Well, I think the task panel should be something like this: for angle, only a button to activate the face reference mode; for offset, only a button too, that only appear when the wall position is selected "offset" (as seen in the video). Maybe whe can change the word "Face" to another or to only a icon (like the icon of face selection filter standard button icon).

[image]
[image]

And changes throughout the user selections:

https://github.com/user-attachments/assets/a0a1555e-b593-403f-ba4a-64c7550c3cb3

pierreporte, 2025-01-06:
> @sheetmetalman , regarding what @pierreporte said, I think that if a user selects a face of another object and then select an edge of a sheetmetal object then in this case you can assume that the user intended for the up-to-face function. No need to select that explicitly.

Yes exactly. But even when the feature has started and an edge is selected, if you select the face or plane, it should do the same (for better UX).

> Perhaps it is better to have a separate command for it, as the addwall, have too many parameters already. With this new tool we can restrict the selection to a single edge and single face.

Disagree. Fundamentally it’s the same feature so a unique command feels legitimate. As @sheetmetalman wrote, I think that this is the best way to achieve it:

> I think this types of face/vertex/edges selection is basic for this type of things (defining lengths, angles etc. Like the in PartDesign Pad/Pocket tool they have "UpToFace", "ThroughAll" and [...]

sheetmetalman, 2025-01-08:
Thanks!

I will look at how to implement such kind of automation (something like select a face before the AddWall command and it gets that face for angle and offset reference).

sheetmetalman, 2025-01-12:
Hi!

Implement that automation (selecting a face before activate the AddWall command) as seen on the video.

https://github.com/user-attachments/assets/8a4db7a4-b835-41e7-8b43-88798d60660b

Also changed the code to user cannot select face for wall creation anymore. It's compatible with old files (where the user has selected a face for wall creation), what I made is only to filter at the time of activating the AddWall command and on the taskpanel. Removing this possibility no need of supplementary angle anymore (but I keep that property for old files and if an error happen, but it isn't appear on the task panel).

Also change the taskpanel to fit better with theses new functions. Replace some checkboxes with checkable buttons and create new icons for better understand what the feature is doing.

https://github.com/user-attachments/assets/a6889328-1386-4b42-9cef-751ecfcd3afb

[image]

The "reverse the wall", "unfold the wall" and "face reference" icons I used pre-existing files from either the main FreeCAD or SheetMetalWB.

pierreporte, 2025-01-12:
@sheetmetalman I’m really impressed by your work!

Not sure about inner sharp mode. It can’t think of anything useful with it. In any case, the icon is unreadable at small size. Maybe you can take inspiration from Solidworks modes: https://help.solidworks.com/2021/English/SolidWorks/sldworks/HIDD_FEAT_SM_EDGE_FLANGE.htm

sheetmetalman, 2025-01-12:
@pierreporte I glad you like it! Thanks!

About the inner sharp it's already on the AddWall command, I only made the icon. But it's like the "inside bend size", on the 90° angle it's more easy see the use of this mode (when you want to define the internal size of the flange).

Indeed! That icon is something to improve! :sweat_smile: This icon was really hard to find a better, but in SolidWorks it's almost unreadable too. SolidWorks call this mode as "Inner Virtual Sharp".

[image]

pierreporte, 2025-01-12:
You’re right. Hopefully the future icon theme will make it easier to read.

FYI I’ve asked the design working group for their opinion on your UI/UX proposal.

hyarion, 2025-01-12:
Just an idea, take it or leave it;
Instead of buttons with icons, how about using something similar to how PartDesign's Pad tool has a Type with a Combobox? QComboBox also supports having an icon before the label:
* https://doc.qt.io/qtforpython-5/PySide2/QtWidgets/QComboBox.html#PySide2.QtWidgets.PySide2.QtWidgets.QComboBox.addItem
* https://www.geeksforgeeks.org/pyqt5-how-to-set-icon-to-the-each-item-in-combobox/

sheetmetalman, 2025-01-12:
@hyarion Thanks for the idea!

I think it's a good approach when you have many options to select or if it isn't a main thing of the feature. I think it will be good for selecting the offset type on this case, 'cause it's not used much.

But on the other hand, with few options and when it's very used and main option, I think the buttons is better, because it need less click to user change the option.

In the case of selecting the "length type" and the "wall position" (which are widely used options in this resource), with buttons user need 50% less clicks to change his option.
With combobox, the user must click on the arrow and after click on the choice: 2 clicks. With buttons the user just click on the button: 1 click. Easier.

pierreporte, 2025-01-12:
It’s also a matter of consistency across FreeCAD. In general (I mean not necessarily for this feature), it’s always a good idea to have a consistent user experience and similar interfaces for similar tasks.

sheetmetalman, 2025-01-12:
I agree!

kadet1090, 2025-01-12:
There are two things in favor of using QComboBox (which is select box in reality) here:

1. Consistency - most "Mode" selectors in FreeCAD are done as Select Boxes as they are easy to extend if there comes a need for more options. 
2. Select Box brings labels for the operations, which makes them easier to understand especially for first time use. Icons should be only used as additional aid, not the only thing to differentiate between modes.

Great work with the icons btw, they are really good!

It also would be good to not only calculate the angle but also to store the face that is used for the angle so if the angle of the face changes for some reason the sheetmetal would automatically update. As others have mentioned Up To Face UI is a good example here to follow so the program will be consistent.

sheetmetalman, 2025-01-12:
@kadet1090 Thanks!

Well then, I will think better about using ComboBox or buttons.

The reference face for angle and offset are sublink properties.

shaise, 2025-01-13:
Awesome work @sheetmetalman ! 👍 👍

sheetmetalman, 2025-01-17:
I tried here to use ComBox instead of buttons for Length Spec and for Wall Position options.

The icon size of selected option is fine. The problem is the size of the item in the list. As seen in the video below.
I don't find a way to adjust this part. Someone know how to do that?

https://github.com/user-attachments/assets/d53b601a-dded-4587-a355-3259077115b7

[image]

sheetmetalman, 2025-01-17:
@hyarion Thanks for answer!

The code is on [here from line 2265 onwards](https://github.com/sheetmetalman/FreeCAD_SheetMetal_ManFork/blob/master/SheetMetalCmd.py#L2265) and [here](https://github.com/sheetmetalman/FreeCAD_SheetMetal_ManFork/blob/master/Resources/panels/FlangeParameters.ui).

I tried this `self.form.LengthSpec.setIconSize(QtCore.QSize(32, 32))` on the `SheetMetalCmd.py`, but it only change the size of the icon after the option is set.

hyarion, 2025-01-18:
This is really strange. This works when I run this in my qt6 build environment (qt6) (`pixi run bash` -> `python`)
```py
from PySide6 import QtCore, QtWidgets  # use PySide without 6 when inside freecad

class Foo(QtWidgets.QWidget):
    def __init__(self):
        super().__init__()
        self.form = QtWidgets.QVBoxLayout(self)

        self.combobox = QtWidgets.QComboBox(self)

        # Use standard icons for stand-alone-liness
        app_style = QtWidgets.QApplication.style()
        items = [
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_FileIcon), "File"),
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_DirIcon), "Folder"),
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_MessageBoxWarning), "Warning"),
        ]
        
        for icon, text in items:
            self.combobox.addItem(icon, text)

        self.combobox.setIconSize(QtCore.QSize(40, 40))
        self.form.addWidget(self.combobox)

app = QtWidgets.QApplication([])  # skip line when inside freecad
window = Foo()
window.show()
app.exec()  # skip line when inside freecad
```

But when I run the same thing inside FreeCAD, both qt5 and qt6, (note comments, and if pasting it, skip all empty lines) it doesn't work. Then I get your behavior where it only shows with the size in the drop down list, not in the non-expanded widget.

The following can be tested by pasting it in the python window in freecad:

```py
from PySide import QtCore, QtWidgets
class Foo(QtWidgets.QWidget):
    def __init__(self):
        super().__init__()
        self.form = QtWidgets.QVBoxLayout(self)
        self.combobox = QtWidgets.QComboBox(self)
        # Use standard icons for stand-alone-liness
        app_style = QtWidgets.QApplication.style()
        items = [
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_FileIcon), "File"),
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_DirIcon), "Folder"),
            (app_style.standardIcon(QtWidgets.QStyle.StandardPixmap.SP_MessageBoxWarning), "Warning"),
        ]
        for icon, text in items:
            self.combobox.addItem(icon, text)
        self.combobox.setIconSize(QtCore.QSize(40, 40))
        self.form.addWidget(self.combobox)

window = Foo()
window.show()
```

I suspect stylesheet configuration is messing this up.

hyarion, 2025-01-18:
I just tested it in Classic (can be selected from Preferences) and there it works.

@MisterMakerNL any ideas what it could be?

MisterMakerNL, 2025-01-18:
Stylesheet apply a checkmark to set the active item it might have conflicts with the icon. There is also a similar issue open with the workbench selector icons not having padding. It looks pretty similar.

sheetmetalman, 2025-01-18:
With the Classic theme the icons are the same size in the list and in the no-expanded widget. Even when the icon size is set with the QtCreator Design.

https://github.com/user-attachments/assets/d9e20cc3-ac01-4824-862b-b6f15d2a4882

shaise, 2025-01-18:
It looks great! hope the skin issue will be solved

pierreporte, 2025-01-18:
@sheetmetalman Could you reorganise the task panel in a more logical and easy to read way? The current one is organised more clearly with one input per line with labels for everything, but it has room for improvement. Here is a proposal:

[image]

The offset input is shown only if relevant wall position is selected. I’ve put outer sharp the default leg mode. I couldn’t put icons easily on this mockup but all the buttons are the same as in your last video.

sheetmetalman, 2025-01-19:
> [@sheetmetalman](https://github.com/sheetmetalman) Could you reorganise the task panel in a more logical and easy to read way? The current one is organised more clearly with one input per line with labels for everything, but it has room for improvement. 

@pierreporte Of course!

I think is better to keep the group around the "Angle" and around the "Wall position" because, when the face reference mode or the offset is on, will be more easy to understand.

[image]

[image]

pierreporte, 2025-01-19:
Here is a new proposal. I chose to separate better the cases when the user wants to enter parameters manually and when they want to follow a plane. In this case, only one selection is needed.

[image]

[image]

sheetmetalman, 2025-01-19:
@pierreporte 
Thanks for suggestion!

But I think this new approach proposal make the UI bigger. The UI must be more simple IMO. Maybe for other scenario it can be good.

In addition, it's already on there, if the user want to insert the angle manually, the user input the value of the angle. If the user want a surface for reference, select the the button for face reference.

I think the vertex/face/edge selection to defining lengths/angles/etc must be separateds for each input (maybe on another Issue, we can create a selection of geometric reference for the length of the wall, but this is another history). Also, this automation of selecting a reference face for the angle and for the offset before triggering the AddWall command is already implemented.

pierreporte, 2025-01-19:
> But I think this new approach proposal make the UI bigger. The UI must be more simple IMO. Maybe for other scenario it can be good. In any case it’s not too big.

It’s only one line longer than the biggest image you posted in https://github.com/shaise/FreeCAD_SheetMetal/issues/307#issuecomment-2600489532. By putting the two radio button on the same line, the height should be the same or even shorter because there are no frames.

> In addition, it's already on there, if the user want to insert the angle manually, the user input the value of the angle. If the user want a surface for reference, select the button for face reference.

> Also, this automation of selecting a reference face for the angle and for the offset before triggering the AddWall command is already implemented (in this new code for face reference).

The idea was to provide the same ease of use if the user hasn’t selected geometry before. (Also there’s a big UX issue right now because the user always need to select geometry first.) For me, entering the angle while specifying an offset to a plane (or the opposite) doesn’t really make sense.

> I think the vertex/face/edge selection to defining lengths/angles/etc must be separateds for each input (maybe on another Issue, we can create a selection of geometric reference for the length of the wall, but this is another history).

It is true that even the length would need this mechanism, but it can be decoupled from angle and position. BTW I plan to open a new issue to request the ability to make flanges from a sketch, which would combine all 3 (position, angle, and length + shape).

----------------

We can have your last UI proposal, it’s fine even though it could be refined further as propose. Also it could be done later.

kadet1090, 2025-01-19:
> But I think this new approach proposal make the UI bigger. The UI must be more simple IMO. Maybe for other scenario it can be good.

This is not a bad thing though. Trying to fit too much stuff in small space does not result often in simple UI but in very crammed one that is hard to navigate. 

[image]

For example in case of your screenshot the marked boxes are not labeled and so it hard to guess what they represent and they are very narrow to a point where they are not able to contain the full name of referenced feature - this is no good. The same goes for other non-labeled inputs, we really should avoid that as it makes UI harder to read - I for example have no idea what `-100.28mm` means looking at the screenshot and I should be able to understand it.

Another thing is that the double border that is created by using non-labeled nested QGroupBox should be avoided - if you want to group related things together that is already part of the group box it would be better to simply use a bit of spacing in-between.

I think that proposal from @pierreporte is a very good one and fits well with the direction that we are aiming for the core of FreeCAD.

sheetmetalman, 2025-01-19:
> (Also there’s a big UX issue right now because the user always need to select geometry first.)

The user don't need to 'always' select the geometry first. When the task panel is open, user can select the geometry reference too (just click the button).

> It is true that even the length would need this mechanism, but it can be decoupled from angle and position. BTW I plan to open a new issue to request the ability to make flanges from a sketch, which would combine all 3 (position, angle, and length + shape).

Agree 100%! Indeed it's a very needed feature at SheetMetalWB.
The AddWall command already has a property to do this, but it is need to be improved. Sometimes, it don't work as expected. Also it's not in the task panel. The property is called "Sketch". I attach a file as example, where it's used.
Perhaps this capability needs to be separated into another command altogether. As we see in the SolidWorks "Miter flange" and "Swept flange" commands, which are two separate commands from the "Edge flange" command.

[miter flange.zip](https://github.com/user-attachments/files/18469790/miter.flange.zip)

sheetmetalman, 2025-01-19:
> We may not be discussing about the same thing. With the latest workbench release, the `SheetMatel_AddWall` command is greyed out until you select an edge. You can’t activate the command first and select the edge later like fillets in PartDesign.

Probably! Sorry. I don't look for this yet.

> I had in mind something like the following image, but your example shows something that also need an to be accessible to the user. In the image, the sketch would become a flange with a cutout. But let’s wait for the issue to discuss about the details.

Now I got the idea! Indeed too! But this is something to be part of this command AddWall, IMO. Something like each wall created by the command has it's own settings (sketch, radius, angle, k-factor etc). Similar to the image below.

[image]

kadet1090, 2025-01-19:
> Only if the user select the angle or select the offset the inputs will appear. And it's very simple: if I select the face reference it's very clear that appear the face name. If I select the offset mode is very clear to understand the spinbox that appear after this is to put the offset value.

I get that this will probably be quite easy to get when used, but we really should avoid users having to guess what value means - label should always be available to consult what the field does mean. It might be easy to understand at first but is also easy to forget.

In the video I also see that when you click checkbox the whole row disappears - this should not happen. When clicking checkbox my instinct is that I'll be able to re-check it if needed and it disappears on click which will cause confusion. If the function of checkbox is to clear the selection - it should be clearly labelled button. In that case it probably should be in the same line as plane selection.

sheetmetalman, 2025-01-19:
> I get that this will probably be quite easy to get when used, but we really should avoid users having to guess what value means - label should always be available to consult what the field does [...]

Now I see!
And another thing is that if the user opens the feature in the future, they will need to know what this value means.

> In the video I also see that when you click checkbox the whole row disappears - this should not happen. When clicking checkbox my instinct is that I'll be able to re-check it if needed and it [...]

Got it! I'll improve this checkbox thing! Looking at the video from this perspective, it does seem strange.

pierreporte, 2025-01-19:
@PaddleStroke Any idea how to make this work with new datum planes?

PaddleStroke, 2025-01-20:
The datum planes are basically just a App.Placement. More specifically, a datum plane is the XY plane of it's placement property.

I don't know sheet metal wb, but the placement is probably all you need in order to define the fold's position.

sheetmetalman, 2025-01-30:
Thanks, @PaddleStroke ! I will start now with the datum planes on my code.

Sorry for the delay!

Fixed that issue with the checkbox button and improved labels.
I removed the checkbox - now is only the button, when face reference is off and the user click the button, it will on the face reference mode and allow to user select a face. If the face reference mode is on and the user click the button, the face reference mode just turn off.

Next step is make it work with datum planes.

I was thinking if instead of group for the wall position specs and the bend angle, will be better use only a horizontal line. I tried it here, but he horizontal line don't appear on the task panel in FreeCAD.

https://github.com/user-attachments/assets/c0814436-23a2-48e4-8962-19d73faa6984

sheetmetalman, 2025-01-31:
About the horizontal line: finnaly I made it. Instead of insert a Horizontal Line directly, I had to insert a layout into Qt Designer and morph into a QFrame, after this change the frameShape setting to HLine - then it works (the horizontal line appeared in the FreeCAD task pane).

I think the horizontal line looks better than the groupbox here.

https://github.com/user-attachments/assets/b06a51e0-8b77-4bdf-bc19-e253a04e2d57

pierreporte, 2025-01-31:
@sheetmetalman It looks really good. The relative angle and material position could be on separate lines, but it’s great already.

luzpaz, 2025-01-31:
Looks awesome!

sheetmetalman, 2025-01-31:
> [@sheetmetalman](https://github.com/sheetmetalman) It looks really good. The relative angle and material position could be on separate lines, but it’s great already.

@pierreporte Done! 

[image]

pierreporte, 2025-02-01:
Now it's perfect.

The only missing thing is support of datum planes, right?

sheetmetalman, 2025-02-01:
The current status is this:

LIMITATIONS:
* Works correctly only with Faces that are Plane or BSpline approximation of a plane.

KNOWN ISSUES:
* Don't work with datum and origin planes.
* Support to work with more than one angle axis is missing. The need is: angle fine tune and offset fine tune (the fine tune is needed because of bend radius and offsets modes adjustments).

The second issue is when you have a reference face which has more than one axis rotation relative to the sheet metal.
At the moment, on these cases, the offset is calculated correctly, but miss the fine tune (because the bend radius and thickness calculation don't work properly with more than one angle - a "3D angle"); and the angle calculation uses only the angle of one axis, the other angles are disregarded.

[image]

[image]

sheetmetalman, 2025-02-01:
I was trying to fix that 3D angle thing here and it looks like I made it.

Now, the current status is this:

LIMITATIONS:
* Works correctly only with Faces that are Plane or BSpline approximation of a plane.

KNOWN ISSUES:
* Don't work with datum and origin planes.

https://github.com/user-attachments/assets/e4422d83-f9ef-4d03-82b5-95d390b4b169

shaise, 2025-02-02:
Great progress!

sheetmetalman, 2025-02-02:
Working with datum planes now!!! :smiley: :partying_face:

But don't work with the current origin planes. :expressionless:
Maybe with the PaddleStroke new feature for the origins planes it will works. When possible I will try it.

Current status is:
LIMITATIONS:
* Works correctly only with Faces that are Plane or BSpline approximation of a plane.

KNOWN ISSUES:
* Don't work with current origin planes (FreeCAD version 1.0.0).

https://github.com/user-attachments/assets/0351626c-3d1a-4e72-b02d-3895bcf83773

[14 more replies]

closed 2025-02-10 by pierreporte