shaise/FreeCAD_SheetMetal #311: Error in "make wall"

free777cat, 2024-03-17:
Unexplained error.
[image]
[image]
[examp.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/14628140/examp.zip)

free777cat, 2024-03-17:
This is the second project with such an error

shaise, 2024-03-18:
@jaisekjames ,
Can you take a look at it? seems an issue with auto mitering.

free777cat, 2024-03-22:
What do you mean?

free777cat, 2024-03-22:
Maybe it's these labels?
[image]
[image]

shaise, 2024-03-22:
@free777cat ,

> What do you mean?

I just asked the original developer of this feature to take a look at the bug. Anyway I took a look myself and I think I know what the problem is. A fix is on the way.

free777cat, 2024-03-22:
This is very good. This Error forced me to manually correct very complex details.

shaise, 2024-03-22:
OK, I believe it is fixed now. please update the workbench and recompute your model.

free777cat, 2024-03-23:
[image]
[image]
[image]

Many thanks!!!
Couldn't figure out what the lilac labels are.
Maybe you know?
And then let's close it...

free777cat, 2024-03-23:
Had to uninstall and reinstall. The update didn't work. That's the only way everything turned out well.

free777cat, 2024-03-25:
lilac labels appear when the object has been used in a Boolean operation

shaise, 2024-03-25:
What do you mean?

free777cat, 2024-03-26:
Sorry. This is a false statement.

free777cat, 2024-03-28:
Houston! We have problems again!
Please look at this ((
[Uploading examp1.zip…]()

free777cat, 2024-03-28:
[image]

free777cat, 2024-03-28:
[examp1.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/14792291/examp1.zip)

shaise, 2024-03-28:
Yes you are correct. 
This is limitation of sheetmetal wb - fold works good only on square faces.
This will work:
[image]

And this will not work:
[image]

shaise, 2024-03-28:
hi @free777cat 

I have enhanced the bend function to work on refined faces as well. It should work now in most cases.
shai

free777cat, 2024-03-29:
Thank you very much!!!!!!!!!!!
An important new project was in jeopardy.

free777cat, 2024-03-29:
I propose to introduce a new element "workpiece" into the "history tree".
This is a virtual element containing the following values: thickness, bending radius, workpiece orientation (in the "sketch" plane or perpendicular).
+possibly sheet sizes.
For further imitation.
It's a lot of code work, but at the beginning I already have this data but it's not marked as "data" in one specific place.

free777cat, 2024-03-30:
Just closed and again a strange error ((
short video
https://youtu.be/BwsipMRXQn8

[image]

free777cat, 2024-03-30:
[examp2.zip](https://github.com/shaise/FreeCAD_SheetMetal/files/14811276/examp2.zip)

shaise, 2024-03-30:
Yes I noticed that sometimes auto mitering does not work. I will take a look at it, But in this case you should switch off auto mitering because if you flatten the shape you get an invalid sheet.

free777cat, 2024-03-30:
watch the video, everything changes there

shaise, 2024-03-30:
OK, I found the issue.
When auto mitering you must set the max extend value. the default is 5mm. This is good for 2mm sheet but not 5mm.
Just change "Max Extend" value to a higher value if you want the mitering to work:
[image]

closed 2024-04-20 by free777cat