shaise/FreeCAD_SheetMetal #317: LCS  in "Add base shape"

free777cat, 2024-03-23:


free777cat, 2024-03-23:
[image]
[image]

it will be more convenient 

+ 
[image]

free777cat, 2024-03-26:
Please change the LCS position.
It interferes a lot with the design.

free777cat, 2024-03-26:
This LCS position nullifies all the advantages of using the "Add base shape" function

shaise, 2024-03-27:
I do not understand your request. Some people like the LCS on the side, some in the center. If I put it in the center someone else will say it is better on the side.

free777cat, 2024-03-27:
When designing, the base of symmetrical parts is always in the center.
Thus, the model can immediately be used in "Assembly".
It's not just "people like", it's an engineering way.

free777cat, 2024-03-27:
If the "Bend side" parameter is added, it does not correlate at all with the lateral position.

[image]

shaise, 2024-03-27:
I can understand the reasoning of putting LCS in the symmetrical center.
But from your image you want it to be at the Celter-Left, What about Center-Center or Center-Right?
[image]

shaise, 2024-03-27:
OK. Best thing is to let the user choose origin location. I have added this option. 
It should also work on previously generated base shapes, it you recompute them.
[image]

free777cat, 2024-03-28:
You still need to add "invert".
Then all options will be available.

[image]

shaise, 2024-03-28:
What do you mean by invert? so the u shape is along Y instead of along X?

free777cat, 2024-03-28:
[image]
[image]
[image]

shaise, 2024-03-28:
Sorry but I do not see a reason for that. You can just rotate the part. This will just over complicate the tool.

free777cat, 2024-03-30:
Is it possible to make the "negative" value work?
"invert"

[image]

free777cat, 2024-03-30:
location ICONs super !!!

closed 2024-04-20 by free777cat