shaise/FreeCAD_SheetMetal #321: DWG Review:  Create Base Shape Panel

sliptonic, 2024-03-31:
Concept panel and tool

[image]
[image]

shaise, 2024-03-31:
This function already have a task panel. comments are appreciated:
[image]
note that there is no need for an apply button as every change in the fields of the panel are interactively shown on the part.

maxwxyz, 2024-03-31:
My remarks: maybe add an option without a bend. Just to be able to create a flat sheet without the need for a sketch.
Also both tools seem to be called base shape (both issues have the exact same name @sliptonic) https://github.com/shaise/FreeCAD_SheetMetal/issues/328
Maybe this tool should keep the base shape name (or something like *default/standard presets*) and the other could be renamed to *Convert to Sheet* as it converts wires or closed sketches to a sheet.

shaise, 2024-03-31:
Definitely agree for both remarks!

maxwxyz, 2024-03-31:
Also, currently you create the feature and need the property editor to change parameters.
For all the issues and task dialogs, the idea is that you can double click on the feature in the tree and that same dialog opens and you can adjust the features in the task dialog and save changes.

shaise, 2024-03-31:
Yes. Exactly as base shape is done now. Double clicking on it in the tree reopens it and let you make changes.

shaise, 2024-03-31:
> My remarks: maybe add an option without a bend. 
Added

shaise, 2024-03-31:
Perhaps we need this dropdown to be images instead of text?

maxwxyz, 2024-03-31:
I'm ok with the dropdown, they are descriptive and I think images would be too small on the dropdown scale for such complicated shaped (e.g. hat). Plus it's interactive so you can just change it and see the result.

MisterMakerNL, 2024-08-24:
Ok not going to open a new ticket for this but would you be so kind to turn the grid of Qpushbuttons in a grid of Qradiobuttons, that would be waaaaaaaaaay better?

pierreporte, 2025-01-14:
@shaise Can we close this?

closed 2025-01-15 by shaise