shaise/FreeCAD_SheetMetal #322: DWG Review: Create Stamp Task Panel

sliptonic, 2024-03-31:
Concept panel and tool

Like ‘Create Flange’, this task is designed to remain open and allow for continuous creation

[image]
[image]

shaise, 2024-03-31:
I think this tool panel should eventually have an x,y grid repetition option as in many cases stamps are repeated in grids.

maxwxyz, 2024-03-31:
@shaise or could the tool also work with the pattern tools of the Part Design WB?

shaise, 2024-03-31:
It can definitely work with the one of the part design, I was just wondering if this is a rather common scenario, maybe its worth doing it in a single step.

pierreporte, 2025-03-09:
@shaise I think this can be closed since commit 9c020ff1a3f94db13ed886233f014e4216e88c76. The follow-up is #305.