shaise/FreeCAD_SheetMetal #326: DWG Review: Create Flange Panel

sliptonic, 2024-03-31:
Concept panel and tool

This is the most complicated Task Panel in the workbench due to the high number of parameters involved. Emphasis was placed on clarity, space, and managing complexity. This will likely serve as the template for new guidance written to FreeCAD’s design guide. 

If possible the ‘Advanced Flange Parameters’ QSInt rollup should be collapsed by default when the task is initiated.

It is also recommended that the task be persistent, allowing for the rapid creation of multiple consecutive flanges until the user is ready to move on to other functions.

[image]
[image]
[image]

shaise, 2024-03-31:
I'm not sure the term Flange is good for that. Usually the flange is on the lip of the part and the last bend created. I do not know if Bend is the correct one, but Flange might be misleading

maxwxyz, 2024-03-31:
So maybe https://github.com/shaise/FreeCAD_SheetMetal/issues/325 Extend could be Extend flange or add flange and this here should be just *Fold* or *Bend*?

shaise, 2024-03-31:
I do not know if extend is a flange as well. Extend is used to make the walls wider to reduce gaps.
As far as I understand this is a standard box:
[image]

and this is a box with a flange:
[image]

So the walls are bends, and flange the protruding lip. Both operations uses "Bends" but only the last one is considered as a "flange". We need an expert machinist remark on it.

pierreporte, 2025-01-14:
@shaise Can we close this?

shaise, 2025-01-15:
Indeed. Closing

closed 2025-01-15 by shaise