shaise/FreeCAD_SheetMetal #327: DWG Review:  Rename 'cutout tool

sliptonic, 2024-03-31:
[image]

shaise, 2024-03-31:
Definitely should be renamed. Also to minimize workflow confusion, we should not show the pocket tool of the part design.

maxwxyz, 2024-03-31:
Is there a way to specify cutting depth currently? I can remember I had the problem once, that it did not cut through the entire sheet thickness. Maybe a user also does not want cut through all.

shaise, 2024-03-31:
I think that in sheetmetal, the cut should always be through all. Otherwise it will not be a part that can be laser cut from a flat sheet. If, for any reason the user wants to make such operation, he can always do it with the part design pocket feature.

pierreporte, 2025-01-14:
@shaise The change has been made so I think the issue can be closed.

shaise, 2025-01-15:
Right now we have both tools. I plan to phase out "sketch on sheetmetal", so, I believe we can close it

closed 2025-01-15 by shaise

pierreporte, 2025-01-15:
What do you mean by phasing out the feature? Will there be one to make cutouts in unfolded state instead?

shaise, 2025-01-15:
There are 2 tools now: 
1. Sketch on sheetmetal: this is essentially a duplication of the PD pocket tool. 
2. Extruded cutout. The new tool.

Both tools are quite similar - however the first tool, when cutting a diagonal flange will create diagonal cuts - which creates a non 'legal' sheetmetal part, and the new one ensures the the cuts are always 90 degreed to the sheet plane.
I want to remove the first tool. It will still stay in the code so old files can be loaded, but will not show on the toolbar

pierreporte, 2025-01-15:
But it will remove the ability to wrap the cutout around the bent part, as if the flat sheet was cut and then bent. The ability to do that shouldn't disappear IMO. It could be a good idea to integrate it into the new tool, with an option to cut the bent part or the flat sheet, while allowing sketches everywhere and not only a face for both modes. An alternative way would be to make un unfolded part a status that can be on and off, to allow user to manually chose (as in Catia).

shaise, 2025-01-15:
Oh, I just found out what yo mean... sorry about that, I was under the impression that the hole is just projected. the sketch actually curves with the bends. 
You are right. Its two different tools. They will both stay. 
So back to square one. do we need to change the name?

shaise, 2025-01-15:
@sheetmetalman , Yes, I think we only need to rename it to Wrap Cutout and perhaps change the Icon

pierreporte, 2025-01-16:
Renaming to wrap cutout is a good idea. It’s more descriptive.

Merging the two tools is also a good idea, I even suggested it, because it will allow for sketches that are not on a face (the user would still need to select one though). However it can be done later.

@sheetmetalman You should open an issue for these bugs.

shaise, 2025-01-21:
I agree completely

sheetmetalman, 2025-01-22:
> I like wrap cutout, too, but I would like to keep the tools separate as they are two completely different tools.

I agree too.