shaise/FreeCAD_SheetMetal #329: DWG Review:  Proposed 'container' workflow

sliptonic, 2024-03-31:
[image]
[image]
[image]

maxwxyz, 2024-03-31:
In addition, maybe this could be overhauled in the future with the material redesign. SheetMetal properties like e.g. k-factor could be set in the material and the SM features could default to the material properties of the given material.

pierreporte, 2025-01-12:
@shaise What’s your opinion on this?

shaise, 2025-01-13:
Right now sheetmetal objects are more or less an extension of PartDesign, and the container is a Body. This means that PD features are intermixed with SM ones. Having a SheetMetal container means we need to restrict it to SM operations only (making sure the model is a 'legal' sheetmetal object). Its an interesting idea, but might be not that easy to implement.

pierreporte, 2025-01-13:
The container could simply be a body with custom parameters, that would be read by the workbench (see picture below). We discussed an alternative way, like Catia does: having a sheet metal parameters object in the tree, which would effectively be a VarSet. It has the advantage of being more visible.

My new file formats proposal would allow for a specific file type for sheet metal (basically a string stored in the document to indicate that the part is a sheet metal part), and FreeCAD would understand that to activate only the sheet metal workbench and other generic tools.

[image]

sheetmetalman, 2025-02-11:
It seems like a good idea!

Just like `Auto link Bend Radius` preference of SheetMetalWB, each object in the tree that need one of these properties is linked to them (in the body property) through expression engine. So, if needed it can be changed. Just as an example, a pipe clamp might have 2 different radii and 2 different k-factors.

pierreporte, 2026-03-06:
@shaise What would you think of automatically creating a VarSet object, then automatically create expressions in task panels so that the default values are pulled from the Varset?